qmk/qmk_firmware #35: Is it reasonable to "unfork" from TMK, so your code can be searched?

BlindLemonLipschitz, 2015-11-07:
I'm not sure if there are advantages to keeping your repo as a fork of TMK.  [here](https://stackoverflow.com/questions/29326767/unfork-a-github-fork-without-deleting) is a Stackoverflow question concerning how to unfork. When trying to search your repo now I get:
>Sorry, forked repositories are not currently searchable.

It would be very helpful to search trough your code when customizing functions/keymaps

jackhumbert, 2015-11-07:
It's still a fork because we pull updates from TMK - at some point it would be nice to merge upstream, but there are still some things being worked out with that.

BlindLemonLipschitz, 2015-11-07:
Just checking, I thought there must be a good reason. Thanks

closed 2015-11-07 by BlindLemonLipschitz