qmk/qmk_firmware #43: How do i mount a keyboard?

peterobrien, 2015-12-17:
I have just received a nice looking ErgoDox EZ, and now I want to change the layout

```
If you're looking to customize a keyboard that currently runs QMK or TMK , find your keyboard's directory under /keyboard/ and read the README file. This will get you all set up.
```

But how do i mount it to be able to find the /keyboard/ folder? It behaves just like a keyboard when i plug it in? There is no volume mounted etc... I'm on a mac

jackhumbert, 2015-12-17:
Hit CMD+space, type in "Terminal" and that should get you going :)

peterobrien, 2015-12-17:
➜  ~  cd /keyboard/
cd: no such file or directory: /keyboard/

jackhumbert, 2015-12-17:
You need to download the firmware, then cd to that folder, then `cd keyboard/` (leave off the leading slash).

peterobrien, 2015-12-17:
Thanks, making progress.

So I want to change keymap, I have edited `keyboard/ergodox_ez/keymaps/keymap_default.c`. Any tips on how i get that on to the keyboard itself?

tdegrunt, 2015-12-17:
We have some OSX instructions here: https://github.com/ErgoDox-EZ/reactor#mac-osx

closed 2015-12-18 by jackhumbert