qmk/qmk_firmware #48: Strange behaviour when assigning certain key as layer switch

majutsushi, 2015-12-21:
Hi,

I'm currently fiddling with the layout of my new ErgoDox EZ, and noticed something strange when I was trying to assign the big left-most key of the left thumb block (which is assigned to "Space" by default) as a layer switch key. Pressing the key does switch to the desired layer, but it doesn't seem to be able to switch back to the default layer, no matter whether I assign it as a temporary switch (with `MO`) or as a toggle (with `TG`). I don't know if that's a hardware or a firmware issue, but it's a bit unfortunate.

will, 2015-12-21:
Does the key you're using to toggle have a value other than `KC_TRANS` on the layer you're toggling to?

majutsushi, 2015-12-21:
That's a good point, I hadn't thought of that. Indeed it was assigned to a different key. Changing it to transparent fixed it. Thanks!

closed 2015-12-21 by majutsushi