qmk/qmk_firmware #49: Using one key for momentary switching and toggling of layer

majutsushi, 2015-12-21:
For some layers it would be useful if you could switch them temporarily and also toggle them. Since it would make sense with the hold/toggle functionality to assign both of these to one key I tried to do this: `LT(2, TG(2))`. Unfortunately this doesn't work, in fact it makes the key do nothing at all. It would be great if this functionality could be supported.

MatrixManAtYrService, 2015-12-27:
I also am looking to do this.  I'll be tinkering with it in the coming days and will report back if I find a way to pull it off.

MatrixManAtYrService, 2015-12-28:
I also would like to do this.  They're talking about it at post #420 here:

https://geekhack.org/index.php?topic=41989.400

I tried using
   ACTION_LAYER_TAP_TOGGLE(1) 
instead of 
    LT(1, TG(1))
But again I found that the key just did nothing.

MatrixManAtYrService, 2016-01-01:
I got this working, somewhat:

    // the defalut values are shown here, but changing them has no effect 
    #define TAPPING_TERM 200 // tap toggle timeout (ms)
    #define TAPPING_TOGGLE 5 // tap toggle count

    const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
    [BASE] = KEYMAP(  // layer 0 : default                                     

    // some rows omitted                                                       

    KC_FN1, KC_FN2, KC_RGUI, KC_MENU, KC_RCTRL,                                

    // more rows omitted                                                       
    ),

    const uint16_t PROGMEM fn_actions[] = {                                    
        [1] = ACTION_LAYER_TAP_TOGGLE(SYMB),              // FN1 - Tap/Toggle Layer 1 (Symbols)
        [2] = ACTION_LAYER_TAP_TOGGLE(NAV)                // FN2 - Tap/Toggle Layer 2 (Nav)
    };  

I'm not wise in the ways of preprocessor voodoo, so I don't have a good idea of why my changes to tapping-term or tapping-toggle don't stick.

dragon788, 2016-01-02:
@MatrixManAtYrService I think that they also include other files that define those same variables, so I'd look for all the places they are set and see if you can determine the inheritance order.

mecanogrh, 2016-01-18:
Can't we have double tap for this? So hold=momentary, double tap=toggle and tap=you just got a free key!

jackhumbert, 2016-01-18:
@majutsushi You'd have to make a custom macro for that, but it's possible! See #57 for a quick example of using the timers. You'd probably want a counter variable to detect which tap is happening.

fuzzbomb, 2016-02-07:
I noticed that the default keymap for Ergodox-EZ already has this behaviour working for `KC_FN1`, since 2739aa8. I discovered it by accident while using the keyboard. The Egodoz-EZ doesn't actually document this; it says the bottom right key is a temporary toggle only.

It uses an approach like @MatrixManAtYrService suggested on 1 Jan, but without messing with timing.

See https://github.com/jackhumbert/qmk_firmware/blob/master/keyboard/ergodox_ez/keymaps/default/keymap.c

jackhumbert, 2016-06-18:
Support for the chainability mentioned won't happen unfortunately - it'll need to be a custom function with process_record_user. Closing this for now.

closed 2016-06-18 by jackhumbert