qmk/qmk_firmware #54: Trouble building - no rule to make target

Lknechtli, 2015-12-22:
When I run `make clean` then `make KEYMAP=keymap_default` or any other keymap, I get `make: *** No rule to make target 'ergodox_ez.elf', needed by 'elf'.'`
I am running both commands in repo/qmk_firmware/keyboard/ergodox_ez
Tried following the instructions both on linux and windows with the same result - am I doing something wrong?

Lknechtli, 2015-12-22:
Nevermind ... Docs should be updated to reflect that you only need the part of the filename that comes after "keymap_"
It's just plain misleading right now.

ezuk, 2015-12-23:
Thank you, @Lknechtli! I changed the wording now -- is it clearer? I appreciate you opening an issue.

closed 2015-12-23 by ezuk

Lknechtli, 2015-12-25:
Looks good to me! I got the replacement left half of my keyboard in the mail today, so I'll check it out when I get home on sunday.  Expect a PR from me sometime soon :)