qmk/qmk_firmware #57: Running a function on every keypress? (Or: changing the behavior of CTL_T and friends)

gavinwahl, 2015-12-28:
I've found that the default behavior of CTL_T(KC_ESC) doesn't suit me. I press control and then try do my chord before the timeout happens. What I'd like it to instead is do a keydown for CTL immediately, then on keyup for a short press do keyup CTL, keydown keyup ESC. For a long press, it would just do keyup CTL. The useless control press in the short press case doesn't bother me. I've accomplished this with a macro:

```c
        static uint16_t start;
        if (record->event.pressed) {
            start = timer_read();
            return MACRO(D(LCTRL), END);
        } else {
            if (timer_elapsed(start) > 150) {
                return MACRO(U(LCTRL), END);
            } else {
                return MACRO(U(LCTRL), T(ESC), END);
            }
        }
```

This works very well, but now I've discovered another problem. Sometimes I do my whole chord too quickly, and end up getting an ESC press that I don't want. I want to skip the ESC press if either the timeout has passed OR another key has been pressed. Is there a way to run code on every keypress to record that there shouldn't be an ESC press? Or, is there a way to write another CTL_T that has this behavior?

Thanks

gavinwahl, 2015-12-29:
Argh, a problem implementing this as a macro--it gets cleared when switching layers in clear_keyboard_but_mods. So you can't hold down control and then switch to another layer.

DidierLoiseau, 2015-12-29:
I have a similar issue with `SFT_T`: it works, but quite often I press and release it before the timeout and 2 characters are printed instead of the desired shifted character.

Reducing the timeout would probably make it work better but I think it would be better if pressing another key somehow "confirmed" that the control or shift was desired and not the simple key.

dragon788, 2016-01-02:
I ended up moving all my modifiers to outer keys as the "hold" check was interrupting my typing too often. I would also love to see a "am I holding two keys" check implemented, as I actually wondered about doing sticky shifts with hitting both to toggle CAPS or something similar with any other pair of keys to trigger a layer "lock" from momentaries or something similar.

ezuk, 2016-01-03:
@jackhumbert - this idea of "am I holding two keys" is super interesting. Do you think there's an elegant way for us to do this?

Eric-L-T, 2016-01-18:
@jackhumbert @ezuk Maybe you can take a look at the code for Oneshot modifiers and see how they sense that a second key has been pressed?

jackhumbert, 2016-01-18:
The "am I holding two keys" would have to come in it's own feature package - I'm not sure we'd be able to modify the original TMK core successfully in this manner. I've been throwing around the idea of allowing a flag where all of the keypresses get routed into a separate function where people can customise behaviours a little easier. This, along with the original feature request is something that could be done inside there.  

@Eric-L-T It's all right [here](https://github.com/jackhumbert/qmk_firmware/blob/master/tmk_core/common/action.c#L103).

gavinwahl, 2016-01-19:
I've implemented the behavior I want by adding a global keypress_count variable:

```patch
diff --git a/tmk_core/common/action.c b/tmk_core/common/action.c
index 77ea39e..415df98 100644
--- a/tmk_core/common/action.c
+++ b/tmk_core/common/action.c
@@ -34,6 +34,8 @@ along with this program.  If not, see <http://www.gnu.org/licenses/>.
 #endif
 
 
+unsigned int keypress_count = 0;
+
 void action_exec(keyevent_t event)
 {
     if (!IS_NOEVENT(event)) {
@@ -62,6 +64,9 @@ void process_action(keyrecord_t *record)
 
     if (IS_NOEVENT(event)) { return; }
 
+    if (event.pressed)
+      keypress_count++;
+
     action_t action = layer_switch_get_action(event.key);
     dprint("ACTION: "); debug_action(action);
 #ifndef NO_ACTION_LAYER
diff --git a/tmk_core/common/action.h b/tmk_core/common/action.h
index 8a4736d..40f9ca4 100644
--- a/tmk_core/common/action.h
+++ b/tmk_core/common/action.h
@@ -74,6 +74,7 @@ bool is_tap_key(keypos_t key);
 void debug_event(keyevent_t event);
 void debug_record(keyrecord_t record);
 void debug_action(action_t action);
+extern unsigned int keypress_count;
 
 #ifdef __cplusplus
 }
```

And now my macro looks like this:

```c
    static uint16_t start;
    static unsigned int start_keypresses;
    if (record->event.pressed) {
        start = timer_read();
        start_keypresses = keypress_count;
        return MACRO(D(LCTRL), END);
    } else {
        if (start_keypresses != keypress_count || timer_elapsed(start) > 150) {
            return MACRO(U(LCTRL), END);
        } else {
            return MACRO(U(LCTRL), T(ESC), END);
        }
    }
        break;
    }
```

eltang, 2016-04-21:
The native tapping code can detect whether another key was pressed while the key was held. `record->tap.interrupted` is the value to check.

closed 2016-06-18 by jackhumbert