qmk/qmk_firmware #59: Allow overriding TARGET

sethherr, 2015-12-30:
Overriding `TARGET` in the makefiles is important, since it's how the filename is generated and we want to ensure we're not overwriting files in different threads in [Reactor](https://github.com/ErgoDox-EZ/reactor).

Right now, I'm doing this with the `-e` flag. According to GNU make, that's not recommended, so it would be great to not be doing that.

Is there an easy way to only assign a variable in c if it isn't assigned? in ruby it would be `TARGET ||= ergodox_ez`

abperiasamy, 2015-12-30:
Something like this should work. https://en.wikipedia.org/wiki/%3F:#C
x= !x?:y

ezuk, 2015-12-30:
Another idea: What if we actually generated the makefile itself with every build? This would also allow us to potentially specify make options in the future (i.e, use the sleep-mode LED blink or disable it, use NKRO or disable it, etc.)

What do you think

jackhumbert, 2016-06-18:
I think this is covered by the new Makefile changes, which defaults to `$(KEYBOARD)_$(KEYMAP)`.

Re the overriding, `TARGET` is defined in the Makefile, and the line to only assign if it's not been assigned is `TARGET ?= my_name` - as of 1923cff, you can assign a `TARGET` in your Makefile (with `=`) to override the default.

closed 2016-06-18 by jackhumbert