qmk/qmk_firmware #61: Produce another key when combined with shift

Slotkenov, 2015-12-31:
I am looking for a way to produce a different key stroke when pressed in combination with the shift key.

For example: I would like the comma and dot key to behave normally when pressed alone. But when pressed with the shift key held down I would like the keyboard to produce a SHIFT+9 and SHIFT+0 instead (i.e.: the comma and dot keys produce parentheses when shifted, instead of the less than and greater than symbols).

ezuk, 2016-01-01:
Well there's a somewhat crazy way to do this: You could make an entire layer that's a "shift" layer (i.e, sends capital letters and $ and whatnot), and just map those two keys to send ( and ). Then you could remap the Shift key on the keyboard to temporarily toggle that layer on.

In use, this would do exactly what you need -- press and hold "Shift" (which is really a momentary toggle layer switch), then press , and you get (. 

@jackhumbert - is there a simpler way to do this?

Slotkenov, 2016-01-01:
Thanks. If I'm right that would mean though that you would have to define the shifted state for all keys, not just the dot and comma keys, right?

jackhumbert, 2016-01-01:
That's the easiest way I know how! Defining the shift layers is a bit easier with `S()`.

plgruener, 2016-01-06:
I've got the same issue (a hardcoded non-qwerty layout), and a question arises: will there be any disadvantages not having a "normal" Shift-key? Eg. shortcuts not working etc?
(not having my ergodox-ez yet, so cannot test).

jackhumbert, 2016-01-06:
The only odd behaviour I can think of is when you're holding a shifted key, the next key you press will also be shifted, which might cause some confusion when typing symbols quickly.

Eric-L-T, 2016-01-07:
It turns out that someone other than me does notice that! Do you think that it is possible to fix this by redefining `S(key)` to use Oneshot Shift rather than regular shift?

ezuk, 2016-01-07:
@jackhumbert back atcha :)

jackhumbert, 2016-01-07:
That issue is a dup of #50 :)

closed 2016-01-07 by jackhumbert

Eric-L-T, 2016-01-07:
I was referencing your comment:
>The only odd behaviour I can think of is when you're holding a shifted key, the next key you press will also be shifted, which might cause some confusion when typing symbols quickly.
.

jackhumbert, 2016-01-07:
Ah, gotcha - that's really a separate issue. If you wanna create a new issue, we can deal with it there.