qmk/qmk_firmware #62: Ergodox EZ udev Vendor & Product ID are not Teensy compatible

stevendlander, 2015-12-31:
The [udev rules](https://www.pjrc.com/teensy/49-teensy.rules) for the Teensy do not work for the Ergodox EZ, which has a Vendor id of "feed" and a product id of "1307"

As a result, on Ubuntu the Teensy firmware must be run with elevated priviledges (sudo).

DidierLoiseau, 2016-01-04:
I am also using Teensy loader for the Ergodox EZ on Ubuntu (15.10) and I don't have the issue with those udev rules: I can run the loader as simple user and it works properly.

stevendlander, 2016-01-04:
@DidierLoiseau does your machine correctly recognize the teensy vendor and product Ids?

My issue may be a hardware defect with my unit, but I have not confirmed that yet.

DidierLoiseau, 2016-01-04:
I have the same values for vendor and product id, `lsusb` gives:

    Bus 004 Device 033: ID feed:1307  

No device description though.

However as soon as I switch the device to programming mode, it changes as follows:

    Bus 004 Device 035: ID 16c0:0478 Van Ooijen Technische Informatica Teensy Halfkay Bootloader

which indeed matches the udev rules.

Just a thought: you didn't forget to press the "reset" button with a paper clip, right? (c.f. [The Easy Way](https://github.com/jackhumbert/qmk_firmware/tree/master/keyboard/ergodox_ez#user-content-the-easy-way-use-an-existing-firmware-file-and-just-flash-it) to customize your ErgoDox EZ)

stevendlander, 2016-01-04:
I neglected to check lsusb *after* hitting the reset button, but even so my udev was still not honoring the rule file.

I will close this, since it seems to be working for you @DidierLoiseau.

closed 2016-01-04 by stevendlander

dragon788, 2016-01-22:
A bonus of the fact that the Teensy Half-Kay bootloader keeps the standard ID is I was able to set up Virtualbox + Vagrant for a pass through of only that ID, so folks can program their keyboard from within the Vagrant VM. Thanks guys for giving me a hint with the id provided here.