qmk/qmk_firmware #75: Define S() and similar functions to affect only one keypress

jackhumbert, 2016-01-07:
Effectively being:

    MACRO(DOWN(<mod>), TAP(<kc>), UP(<mod>))

Eric-L-T, 2016-01-07:
Will keys that use this still repeat?

jackhumbert, 2016-01-07:
Nope :/ it'll be either-or - repeating is handled on the OS side of things.

Eric-L-T, 2016-01-07:
So there is no way to program keyboards to indicate that keys that use this are held down beyond their initial presses?

Eric-L-T, 2016-01-07:
Or, can the current programming of S() be retained and add something to make Shift release when any other key is pressed?

jackhumbert, 2016-01-07:
That was my motive when renaming the issue - releasing shift on another keypress would have more problems than it's worth if you're worried about key repeating (e.g. $$$$4444 after another keypress).

I think the best solution for this is some sort of firmware/OS tandem solution, but I'll keep this request for a oneshot `S(<kc>)` equivalent open.

Eric-L-T, 2016-01-07:
Well, signifying that a letter keys, whether lowercase or capital, is pretty important when using OS X - it makes a menu for selecting a special version of that letter pop up.

eltang, 2016-04-21:
This issue can be closed. I have already implemented it.

closed 2016-06-18 by jackhumbert