qmk/qmk_firmware #76: custom firmware layout independent of OS-layout

plgruener, 2016-01-07:
(Not a real issue of qmk, but trying to get input/discussion)

My goal: creating a firmware-based layout (of neo2, see www.neo-layout.org ; needs massiv unicode support, but that's not the issue here). That should work independent of the PC I plug my ergodox EZ in.
Question: does the OS-defined Keyboard-layout (whatever this is really called) interfer with this, i.e. do I get different letters pressing the same key under the US and german map?
If yes, what to do?

Eric-L-T, 2016-01-07:
Congratulations on learning an alternative layout! I had this issue as well, and it caused me to think that my keyboard was broken for a moment. Basically, setting the keyboard layout to anything other than the default one in an OS will activate a software built into the OS that will translate any keypress that the OS receives, regardless of the layout of the keyboard that sent it. Therefore, you will have the set the layout in your OS to the default layout if you want to create a firmware-based layout.

jackhumbert, 2016-01-07:
You'll need to either use the US (qwerty) keymap on your OS and program your layout into your firmware, or use your German/Neo2 keymap on your OS and program a standard layout into your firmware - other variations will work, but it'll take some experimentation to figure out!

I don't know what OS you're using, but xmodmap is one way to do it on the OS side.

plgruener, 2016-01-07:
I know I can do it on OS-side (that's how neo currently handles it, via xkbmap), but I want to be able to use the ErgoDox also on foreign computers – where I potentially cannot install any keyboard-drivers.

@Eric-L-T the Problem is: what's a computer's default layout? Here in Germany it's the German (Iso)-Layout, my laptop is set to neo by default.

I suppose all keycodes are based on qwerty (US), so I'd have to use qwerty on all computers?

jackhumbert, 2016-01-07:
It's gonna depend on what locale that computer is set to :/ If it's a US one, it'll be qwerty, and you can use the keycodes at face value. Anything else, and there will need to be some conversion.

You could have a layer for each (much like the qwerty, colemak, dvorak layouts [on the default Planck layout](https://github.com/jackhumbert/qmk_firmware/blob/master/keyboard/planck/keymaps/keymap_default.c#L18)) and switch between them depending on which OS-side keymap the computer is set to.

plgruener, 2016-01-07:
Thanks, that's indeed an interesting Idea. I already wanted to built in an extra qwertz-layer (for games etc). But I'd have 2 layers minimum for each , right? (lower-& uppercase).
I'll test it out as soon as I got my board.

closed 2016-01-07 by plgruener

jackhumbert, 2016-01-07:
Yeah, two per layer sounds right! You have 32 to use, though :)