qmk/qmk_firmware #78: Enable 2 layers at the same time?

DidierLoiseau, 2016-01-09:
I am currently designing a layout in which layer 2 should be the same as layer 1 + some keys.

Layer 2 is designed to be a momentary layer, while layer 1 would be a toggle layer.

To avoid duplicating layer 1 in layer 2, I would like to configure a key that momentarily toggles both layers 1 & 2.

Should Implement it as a macro or is there a simpler way?

DidierLoiseau, 2016-01-09:
I implemented it as a macro and it was quite simple in the end. In `action_get_macro()`:

    switch(id) {
        case 2:
            layer_state ^= (1 << LAYERA) | (1 << LAYERB);
            break;
    }

and use it with `M(2)` in the layout.

I don't check whether it is a key pressed or released since I need to toggle the layer states in both cases.

Anyway I would still be interested to know if there is an alternative without macro.

ezuk, 2016-01-10:
I think there's something I don't understand here. If layer 2 is designed to be mostly the same as layer 1, why not make the keys in layer 2 transparent? I mean, those that should stay the same. Doesn't that work?

DidierLoiseau, 2016-01-10:
That's indeed what I did, see https://github.com/DidierLoiseau/qmk_firmware/blob/typematrix/keyboard/ergodox_ez/keymaps/keymap_typematrix.c

The thing is that enabling layer 2 does not automatically enable layer 1. I thus needed a way to enable both layers at the same time.

jackhumbert, 2016-01-10:
This is a pretty cool implementation! That's the only way I know to do something like this - I can't really see it becoming a core feature right now because of keymap readability concerns.

DidierLoiseau, 2016-01-10:
Ok, thanks. I'm happy to know that's the proper way to implement it and that I didn't miss something.

I don't think this needs to be a core feature either, but it could be useful to document it somewhere.

closed 2016-01-10 by DidierLoiseau

DidierLoiseau, 2016-02-07:
Playing with `layer_state` directly appears to be a bad idea in the end. I quite often end up with keys stuck from the temporary layers.

In fact, it appears I should have used `layer_invert()` instead, which does the proper cleanup if some keys are still held while switching a layer off. This appears to also be the cause of my problem in #81.

EBADBEEF, 2025-05-01:
I ran into a similar issue and searching brought me here. Maybe this will help someone else.

I wanted to turn on and off two layers simultaneously from a single momentary layer keycode. This allows a momentary layer to fall-through to another momentary layer. Working example:
```
enum {
  WB = 0, //win base (fn key is MO(WF))
  MB = 1, //mac base (inherits from win base, fn key is MO(MF))
  WF = 2, //win fn
  MF = 3, //mac fn (want to inherit from win fn layer)
}
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    if (keycode == MO(MF)) {
        const layer_state_t both_layers = (1<<WF) | (1<<MF);
        if (record->event.pressed) {
            layer_or(both_layers);
        } else {
            layer_and(~both_layers);
        }
        return false;
    }
    return true;
}
```