qmk/qmk_firmware #82: Are KC_MEDIA_EJECT & KC_MEDIA_SELECT correct?

DidierLoiseau, 2016-01-10:
Currently trying to reproduce the TypeMatrix 2030 on my ErgoDox EZ.

fn+F6 on the TM is mapped to "eject". I tried to use the `KC_MEDIA_EJECT` for this but it does not seem to send the correct keycode. I am working on Ubuntu, here is the output of `xev` with those keys:

TM 2030:
```
KeyPress event, serial 53, synthetic NO, window 0x5200001,
    root 0xd9, subw 0x0, time 191628480, (142,442), root:(363,710),
    state 0x10, keycode 169 (keysym 0x1008ff2c, XF86Eject), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 53, synthetic NO, window 0x5200001,
    root 0xd9, subw 0x0, time 191628552, (142,442), root:(363,710),
    state 0x10, keycode 169 (keysym 0x1008ff2c, XF86Eject), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```
(note that this works with the default Gnome3 keyboard shortcut – it ejects the CD tray)

ErgoDox EZ:
```
KeyPress event, serial 55, synthetic NO, window 0x5200001,
    root 0xd9, subw 0x0, time 191723071, (585,221), root:(806,489),
    state 0x10, keycode 248 (keysym 0x0, NoSymbol), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 55, synthetic NO, window 0x5200001,
    root 0xd9, subw 0x0, time 191723142, (585,221), root:(806,489),
    state 0x10, keycode 248 (keysym 0x0, NoSymbol), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

Also, I was curious about what was `KC_MEDIA_SELECT` so I added it to my keymap. Here is what it gives:
```
KeyPress event, serial 83, synthetic NO, window 0x5200001,
    root 0xd9, subw 0x0, time 191864282, (301,-52), root:(522,216),
    state 0x10, keycode 179 (keysym 0x1008ff81, XF86Tools), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 83, synthetic NO, window 0x5200001,
    root 0xd9, subw 0x0, time 191864354, (301,-52), root:(522,216),
    state 0x10, keycode 179 (keysym 0x1008ff81, XF86Tools), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```
So Ubuntu recognises it as a "Tools" key. In fact, the default action on Gnome3 for this key is to open the Gnome Settings.

jackhumbert, 2016-01-10:
[It looks like this is a known issue with the eject key](https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#eject-on-mac-osx) - that's all the info I have on it right now :/

DidierLoiseau, 2016-01-10:
Thanks, this is actually discussed in tmk/tmk_keyboard#250

I tried to use F20 as suggested but that does not work: it sends the AudioMicMute. Gnome3 even recognises and apparently mutes the microphone, but I cannot find where this is configured. Anyway I think this is Linux-specific as Freedesktop remapped a Mac Mic-Mute to F20: https://bugs.freedesktop.org/show_bug.cgi?id=54171

closed 2016-01-13 by jackhumbert