qmk/qmk_firmware #85: CTL_T(LGUI(v)) not working

sboesebeck, 2016-01-13:
... acually it produces either a V or ctrl, but not what I expected...

did I miss something here?

The expected behavior would be, holding the key should press CTRL, typing it should send  CMD-V..

jackhumbert, 2016-01-13:
Sorry, both functions are specialised shortcuts to other ACTIONs, and aren't chainable like that.

closed 2016-01-13 by jackhumbert

sboesebeck, 2016-01-13:
ok, so it is not possible to have this functionality at all or is there something possible in future?

jackhumbert, 2016-01-13:
Not with the current implementation - these shortcuts rely on the TMK base, which won't be changed.

You can accomplish this with [a macro](https://github.com/jackhumbert/qmk_firmware/issues/57#issue-124108421) (untested):

        static uint16_t start;
        if (record->event.pressed) {
            start = timer_read();
            return MACRO(D(LCTRL), END);
        } else {
            if (timer_elapsed(start) > 150) {
                return MACRO(U(LCTRL), END);
            } else {
                return MACRO(U(LCTRL), D(LGUI), T(V), U(LGUI), END);
            }
        }

sboesebeck, 2016-01-13:
Thanks a lot for the quick response - I'll give that a try...

sboesebeck, 2016-01-15:
just wanted to let you know, your code works perfectly out of the box!

thanks a lot...

But one question remains: is it possible to toggle a layer in a Macro?

Cheers,

Stephan

DidierLoiseau, 2016-01-15:
@sboesebeck You could play with the `layer_state`, see how I did in #78.

jackhumbert, 2016-01-15:
@DidierLoiseau is right on here - something like this would toggle layer 4:

    layer_state ^= (1<<4)