qmk/qmk_firmware #86: unicode

20lives, 2016-01-15:
tryied using the unicode function to create a hebrew layer keyboard, but when i press one of these keys the keyboard act weird and i need to reconnect it
[keymap_dvorak-heb.txt](https://github.com/jackhumbert/qmk_firmware/files/91979/keymap_dvorak-heb.txt)

have i done something wrong?

jackhumbert, 2016-01-15:
This is still a largely untested feature, but what OS are you trying it on? You'll need to modify [keymap_unicode.c](https://github.com/jackhumbert/qmk_firmware/blob/master/quantum/keymap_unicode.c) to get it to output what your OS needs to unicode character input.

20lives, 2016-01-15:
tryied on osx and windows.
what do i need to modify in the file to make it work on windows?

coderkun, 2016-01-15:
I would also like to know what changes need to be done in order to get it to work with gtk3 under Linux.

jackhumbert, 2016-01-15:
[Check this out!](https://en.wikipedia.org/wiki/Unicode_input#Hexadecimal_code_input)

20lives, 2016-01-15:
so this method is only a alt+(unicode number) macro shortcut?

jackhumbert, 2016-01-15:
Yeap! That's the only way I know of to get this sort of input right now.

coderkun, 2016-01-15:
As far as I understand the action_function() method in keymap_unicode.c has to print out CTRL+SHIFT+U, then the Unicode sequence and finally ENTER (for ISO 14755 compatible systems) as shown below?

        register_code(KC_LCTL);
        register_code(KC_LSFT);
        register_code(KC_U);
        unregister_code(KC_LCTL);
        unregister_code(KC_LSFT);
        unregister_code(KC_U);
        […]
        register_code(KC_ENT);
        unregister_code(KC_ENT);

But with this code, nothing happens.

20lives, 2016-01-15:
cool.

jackhumbert, 2016-01-16:
Yeah, that looks correct - do you have any luck manually entering those keystrokes? I would start there to make sure the OS is ready for the input.

20lives, 2016-01-16:
your function seems to not work or something else but macro i made work great:
```c
const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt)
{
      switch(id) {
        case 0:
        if (record->event.pressed) {
          register_code(KC_RSFT);
        } else {
          unregister_code(KC_RSFT);
        }
		case 128:  
        if (record->event.pressed) {
          return MACRO( I(1), D(LALT),T(P1), T(P2), T(P8), U(LALT), END  );
        }
		case 145:
        if (record->event.pressed) {
          return MACRO( I(1), D(LALT),T(P1), T(P4), T(P5), U(LALT), END  );
        }
        break;
      }
    return MACRO_NONE;
};
 ```

closed 2016-01-16 by jackhumbert