qmk/qmk_firmware #91: Sequence tirggers function

20lives, 2016-01-16:
is it possible to send a keypress after a sequence? for example for a dvorak user who wants to keep the qwerty CTRL functions (copy, cut...) locations like LCTL(KC_K) will result in LCTL(KC_C).

i can't find this feature in the docs.

jackhumbert, 2016-01-16:
I'm not sure what you mean by sequence, but you can use the Dvorak-specific
keycodes like DV_C to get a literal C when the OS is set to Dvorak. You'll
need to include `keymap_dvorak.h` at the top of your keymap.

20lives, 2016-01-16:
read my other example.

i have to use os qwerty

20lives, 2016-01-16:
in kiibohd/KLL the feature available, you can read it in the spec http://input.club/kll.

jackhumbert, 2016-01-16:
I'm still not sure exactly what you're talking about, but if you're looking for a qwerty-ctrl key when your default layout is hard-coded dvorak, a macro like this would work:

```
if (record->event.pressed) {
  register_code(KC_LCTL);
  layer_state |= (1<<5);
} else {
  unregister_code(KC_LCTL);
  layer_state &= ~(1<<5);
}
```

Where layer 5 is the qwerty keymap of the control keys you're talking about, and your control key is `M(x)` (x being the macro number you place this code block into).

Eric-L-T, 2016-01-17:
Do you mind having to do execute all of your shortcuts in QWERTY? If not, here is a solution. You should first create a QWERTY layer. Then you should change the Ctrl keys on your Dvorak layer to function keys that use `ACTION_LAYER_MODS(*insert number of new layer here*, MOD_LCTL)`.

jackhumbert, 2016-01-17:
I believe that's what they're talking about - the macro I wrote does the same thing.

closed 2016-01-17 by jackhumbert

Eric-L-T, 2016-01-17:
Oh, really? Do you know if there is some file in which I can find the macros that correspond to each action?

jackhumbert, 2016-01-17:
Unfortunately macros are implemented in a different way than the actions - [action.c contains a lot of them](https://github.com/jackhumbert/qmk_firmware/blob/master/tmk_core/common/action.c#L209).

Eric-L-T, 2016-01-17:
Oh. I was hoping I could find a solution to the problem I posted in /r/olkb.