qmk/qmk_firmware #96: Macros for mouse T(MS_U) not working.

krasnovu, 2016-01-20:
Hi!
In action_get_macro all of this codes not working:
	return MACRO( T(MS_U), END );
OR
	return MACRO( D(ACL2), D(MS_U), U(MS_U), U(ACL2), END );
OR
	register_code(KC_ACL2);
	register_code(KC_MS_U);

How i can move mouse in max speed without tapping two button?
Thanks!

jackhumbert, 2016-01-20:
That's a good question! I think mouse actions are handled a little differently - the code for that is in [mousekey.c](https://github.com/tmk/tmk_keyboard/blob/5ce9bf8353568b341b79b3cea676ea62a7d40442/tmk_core/common/mousekey.c#L130). If you wanted more control over what's happening, you could include `mousekey.h` (and possibly `host.h` - see the mousekey.h header) at the top of your keymap and try this in a macro (untested):

    
    static report_mouse_t mouse_report = {};
    mouse_report.y = 2;
    host_mouse_send(&mouse_report);

krasnovu, 2016-01-20:
Thanks a lot, Jack!
I am look in mousekey.c and resolve my case by code like this:
```
		if (record->event.pressed) {
			mousekey_on(KC_ACL2);
			mousekey_on(KC_MS_U);
		} else {
			mousekey_off(KC_MS_U);
			mousekey_off(KC_ACL2);
		}	
```

closed 2016-01-20 by krasnovu

jackhumbert, 2016-01-20:
Nice :)