qmk/qmk_firmware #101: How to make.

Woehler, 2016-01-23:
When trying to make my keymap I get the following error "make: *** No targets specified and no makefile found.     stop. "
Do I only need the keymap.c file to compile to .hex or is there supposed to be other files such as a makefile? 
And where exactly am I supposed to put the files relative to the cmd directory?

ezuk, 2016-01-23:
Hi there,

Which instructions are you following? What keyboard are you making for?

Woehler, 2016-01-23:
It the new ergodox ez.

Woehler, 2016-01-23:
pretty much just stuck at step 6. 
Compile your firmware by running make clean followed by make KEYMAP=your_name. For example, make KEYMAP=german. This will result in a hex file, which will always be called ergodox_ez.hex, regardless of your keymap name.

ezuk, 2016-01-23:
Did you perform step 2 of the instructions - clone the repository? You should have the whole repo, not just the C file. 

Can you paste a complete console log of what's happening, and your directory structure?

ezuk, 2016-01-23:
Also, you need to run `make clean` and `make KEYMAP=your_name` from the `keyboard/ergodox_ez` directory, not from within your keymap subdirectory. Is that maybe the problem?

Woehler, 2016-01-23:
Yeah. Followed the steps on how to make from the teensie website. 
Make sense now that I have to make it from that folder. 
However. Now I get an error
 make: Interrupt\Exception caught

Woehler, 2016-01-23:
Thank you. Got it working now. 
Just had to change my directory for WinAVR to fix the last error.
Can finally start making my own layout without using the massdrop configuration.

closed 2016-01-23 by Woehler