qmk/qmk_firmware #105: One key momentary switching and many keys to select the layers

krasnovu, 2016-01-24:
Hello!
How I can do this: Press the key for example that "A" which defined MO(Layer1), hold it and tapping "B" which defined in Layer1 like this MO(Layer2), release "B", but still keep pressed the "A" and now need working Layer2 not Layer1. When I release "A", I returned from Layer2 to base Layer0.
Sorry for my google english.
Thank you!

Eric-L-T, 2016-01-25:
Do you have an issue with holding `MO(2)`?

krasnovu, 2016-01-26:
```
It may seem strange example:

Imagine a small keyboard
/----/----/
| B3 | B4 |
/----/----/
| B1 | B2 |
/----/----/

keymap
{
    [0]={
        { KC_B,    KC_C },
        { HMO(1),  KC_A },
    },
    [1]={
        { SWMO(2),  SWMO(3) },
        { KC_TRNS,  KC_D    },
    },
    [2]={
        { KC_E,    KC_F    },
        { KC_TRNS, SWMO(4) },
    },
    [3]={
        { KC_H,    KC_I },
        { KC_TRNS, KC_G },
    },
    [4]={
        { KC_2,    KC_3 },
        { KC_TRNS, KC_1 },
    }
}

HMO - hold momentary
SWMO - switch momentary

to type 'DB': hold B1, tap B2, release B1, tap B3
to type 'GA': hold B1, tap B4, tap B2, release B1, tap B2
to type '3':  hold B1, tap B3, tap B2, tap B4
```

Eric-L-T, 2016-01-26:
I'm really not sure why you want to set things up this way, but I will tell you how to do it.

In this keymap, `HMO(1)` would be programmed as the following macro:

    if (record->event.pressed) {
      layer_on(1);
    } else {
      layer_off(4);
      layer_off(3);
      layer_off(2);
      layer_off(1);
    }

`SWMO(layer)` would be `TO(layer,1)`.

krasnovu, 2016-01-26:
Eric-L-T, thank you! It is works, but if on key TO(layer,1) in underlying layers define M() or MO() then they triggered too.

Eric-L-T, 2016-01-26:
That shouldn't be happening. Try changing all `TO(layer,1)` to:
    if (record->event.pressed) {
      layer_on(1);

krasnovu, 2016-01-27:
Now works fine, thank you, Eric!

closed 2016-01-27 by krasnovu