qmk/qmk_firmware #108: Change key when pressing SHIF

benoitne, 2016-01-25:
Hi

I just receiving my keyboard and it is almost my first message I am writting with it...
I have built my first custom hex everythinkg is working great on this side !
I am building an AZERTY (I will contrubute here as I don't see any yet)
As AZERTY has different key when pressing SHIFT I am wondering if there is an easy way to customize this?
For example SHIFT comma needs to be questionmark (it is < in QWERTY)
I like a lot they way to produce layout it is very convenient and I am looking for a simple way to be able to add a key:shift key.
many thanks

belette

DidierLoiseau, 2016-01-25:
Are you implementing a _firmware_ AZERTY layout to use with another layout (QWERTY?) on the OS side?

Otherwise you may simply include `keymap_french.h`. Note that I think this one is defined for AZERTY-fr, it won't work properly if you are using another variant like be or ca.

benoitne, 2016-01-25:
Hi Didier
I am trying to create a french AZERTY layout.
I have been working from the default QWERTY layout (keyboard/ergodox_ez/keymaps/default/keymap.c) then I realized that the corresponding SHIFT key are for QWERTY not AZERTY...
On your side how do you do? 
Setting OS in AZERTY or QWERTY?
If I select QWERTY on OS side and including keymap_french.h would be enough to get the right shift key in place?
Many thanks for your help

benoitne, 2016-01-25:
Autoreply!
I keept the QWERTY layout and I have added the keymap_french.h it is all I need to make it work!
Merci

DidierLoiseau, 2016-01-25:
I'm actually using the [BÉPO layout](http://bepo.fr), on OS side. If you are interested in keyboard ergonomics, you should definitely have a look at this French layout. Note that it does not work well with the default ErgoDox EZ keymap, but you could try it with the `tm2030` keymap that I implemented for example.

npoirey, 2016-02-01:
I'm asking here instead of creating a new issue because the title match what I want even if the OP is not looking exactly for the same thing.

I want what the OP want, but i'm planing on having my own disposition.
For instance, I have a key that : 
- when pressed alone, should put '&' symbol
- when pressed with shift, should put '|' symbol
- when pressed with altgr, should put '\' symbol

No problem for altgr, it's just a layer whith function key, but how can I do the change for shift?

Thank you

jackhumbert, 2016-02-01:
The way to do this is through a couple of macros - one to jump to the "shifted" layer, and then one for the `|` key:

```c
const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt) 
{
    switch(id) {
        case 0: // M(0) jumps to shifted layer
        if (record->event.pressed) {
        	register_code(KC_LSFT);
        	layer_on(2);
        } else {
        	unregister_code(KC_LSFT);
        	layer_off(2);
        }
        break;
        case 1: // M(1) is the | key
        if (record->event.pressed) {
        	unregister_code(KC_LSFT);
        	register_code(KC_RALT);
        	register_code(KC_6);
        } else {
        	unregister_code(KC_6);
        	unregister_code(KC_RALT);
        	register_code(KC_LSFT);
        }
        break;
    }
}
```

npoirey, 2016-02-16:
Thanks, took a while because I didn't have the keyboard ready but everything works fine now ! :D

closed 2016-06-18 by jackhumbert