qmk/qmk_firmware #111: No media keys on ErgoDox?

Flink, 2016-01-27:
Hi!

I’m using an ErgoDox (not an EZ one) with this firmware and I found that media keys are not working. They just send nothing.
I’m just using this firmware since yesterday (just received my ErgoDox) so maybe there’s something I missed?

Eric-L-T, 2016-01-27:
What is your operating system? Have you seen [this](https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#media-control-keys-in-mac-osx)?

Flink, 2016-01-27:
I’m using linux. Thank you for the link but the chibios does not apply on an ergodox, does it? And if it is the case how to use it please? (Also it seems weird that the firmware used by the ErgoDox EZ doesn’t send multimedia keys)

Flink, 2016-01-27:
So I read the FAQ and I didn’t see anything about my problem. I may have missed something though.

ezuk, 2016-01-28:
Hey @Flink - so the short answer is I don't know. :) The media keys on the EZ work fine on a PC, and the Mac layout lets people use them on a Mac. I've yet to receive any questions for the EZ re media keys for Linux. 

I'm sorry I'm not able to support just ErgoDox DIY users -- I really wish I could, but if you try running a product like this you'll realize just how much work it is... I'm afraid right now I can only offer personal one-on-one support to ErgoDox EZ customers. It's not because I'm evil, just because of a simple time constraint. I'm sorry and I hope you can understand!

DidierLoiseau, 2016-01-28:
@ezuk I confirm the media keys work fine on Ubuntu with the ErgoDox EZ.

@Flink have you checked the output of `xev` when you press the media keys? Also, you might want to compile and run your keyboard in debug mode with `hid_listen`, it's not very hard to configure.

Flink, 2016-01-28:
@ezuk No problem, I understand perfectly. But this firmware should work without any problem on an ErgoDox DIY no?

@DidierLoiseau Yes I checked and there is no output at all :/ Could you point me towards resources to enable and use this `hid_listen` debug mode please?

jackhumbert, 2016-01-28:
It should - do you have your code forked that we could look at? Have you tried any other OSes?

Flink, 2016-01-28:
@jackhumbert my keymap is here: https://github.com/Flink/qmk_firmware/blob/bepo-flink/keyboard/ergodox_ez/keymaps/bepo_flink/keymap.c
And no I didn’t try with another OS, I’ll try on windows and report back.

jackhumbert, 2016-01-28:
Quick sanity check - your keyswitches work when you're using a different keycode, right? Just wanna make sure it's not a hardware thing.

What linux flavour/environment are you running?

Flink, 2016-01-29:
Yeah keys on the main layer are working as expected and I swapped some keys previously that were also working fine.

I’m running an Archlinux.

Flink, 2016-01-29:
@jackhumbert I just did the test on Windows and the media keys are working as expected. So it seems it’s not working on my arch. That’s strange, could it be a permission problem related to udev or something like that? (my previous keyboard, a typematrix 2030, was sending media keys correctly, I don’t know if it can help)

krasnovu, 2016-01-29:
I have planck keyboard and keycodes like KC_MPLY, KC_MNXT, KC_VOLU... not works in win 7, but works on ubuntu 14.04

Flink, 2016-01-29:
@krasnovu forgot to precise, I tested with Windows 10 and KC_CALC, KC_VOLU, KC_VOLD and KC_MUTE are all working.

Flink, 2016-02-01:
Ok, found the issue, it’s the same as described here: https://bbs.archlinux.org/viewtopic.php?id=178838
Since I don’t use any joystick under linux, I just removed the `xf86-input-joystick` package and now I have working media keys :grin:

closed 2016-02-01 by Flink