qmk/qmk_firmware #112: Mapping a Teensy key

alcesleo, 2016-01-28:
Hello! I'm trying to create a key that acts like the little reset-button on the teensy board, I see this mentioned in lots of places but I can't find anywhere that describes how to do it - could someone please point me in the right direction?

Cheers

jackhumbert, 2016-01-28:
The `RESET` keycode will do just this, as long as your bootloader is set to the right size in your Makefile!

alcesleo, 2016-01-28:
Hmmm, there is something else I'm doing wrong I guess - I'm not sure what the correct size is for the ErgoDox EZ, I tried all the different ones but the button always has the same behaviour - killing the keyboard so you have to plug it out and back in again for any key to work.

jackhumbert, 2016-01-28:
It should be 512 for the EZ, but the behaviour you described sounds exactly like what the reset button does :) does it not show up in the Teensy Loader?

alcesleo, 2016-01-28:
Perhaps I'm asking for the wrong thing, the button I mean is the one you press when you reprogram it, every time I change the layout I do `make clean && make teensy ...`, then after a bunch of log output it halts and asks me to press the reset button, then I stick a little paperclip through the hole in the upper right corner and it moves on. If I program it once using the paperclip (because the button obviously isn't mapped yet) and then again trying it with the button mapped to `RESET` it still just kills all the buttons and the process is still halted. Is there anything else I could be missing?

jackhumbert, 2016-01-28:
That should be the same as the button - I'm not sure why it wouldn't trigger the same signal.

@ezuk is this anything you've seen before?

alcesleo, 2016-01-28:
Same behaviour when using the Teensy GUI, so it doesn't seem like there is anything isolated to using the `make teensy` command

ezuk, 2016-01-28:
hmm, someone else mentioned something similar. Not a hardware issue obviously because the reset button works fine... any way we can trace this in firmware @jackhumbert ?

jackhumbert, 2016-01-28:
[Here's some more info on jumping into the bootloader](https://www.pjrc.com/teensy/jump_to_bootloader.html), which is referenced [here](https://github.com/jackhumbert/qmk_firmware/blob/master/tmk_core/common/avr/bootloader.c) as not working with the LUFA stack (what QMK uses).

[There's also a discussion about it last year](https://github.com/tmk/tmk_keyboard/issues/179) - I haven't been able to try it yet, but it looks like [these changes](https://github.com/technomancy/tmk_keyboard/commit/ebb4f96b074e06be8ab3ee3727ad384f5b78f58b) would work. I'd be happy to merge in those changes with a `TEENSY` flag (saving folks from modifying the bootloader location as well).

mfilej, 2016-02-08:
@jackhumbert I'd be interested in that change :smile:

aspickard, 2016-02-18:
@jackhumbert I was able to test the changes with an original dox and it worked fine.

Rovanion, 2016-02-20:
@aspickard Could you provide a diff for the changes you made? And beyond that I think it's totally fine for you to open up a PR since jack explicitly states that he'd be happy to merge such a change.

mfilej, 2016-02-22:
I'd love to see that diff as well because I'm not confident enough to make the change myself. :smile:

aspickard, 2016-02-22:
@Rovanion @mfilej I'll try to have something up for you guys very soon.

Rovanion, 2016-02-22:
Thank you @aspickard, It's greatly appriciated!

aspickard, 2016-02-22:
@Rovanion @mfilej Here you go! https://github.com/jackhumbert/qmk_firmware/pull/160/files

Lmk if there are any issues/improvements.

Rovanion, 2016-02-23:
Thank you @aspickard, found that what was really missing was a declaration of which controller the ergodox ez is using in it's config.h. As of such the diff can be reduced to what's in PR #162.

I'd be great if some of you could test the change on your device. But it seems to be working on my end.

mfilej, 2016-02-23:
Works for me! Seems #162 is enough but thank you both for sharing the patches :sparkling_heart:

jackhumbert, 2016-06-18:
Closing this as it's been narrowed down to the problem described in #164.

closed 2016-06-18 by jackhumbert