qmk/qmk_firmware #121: Version of LT that also applies a modifier key

jeffschenck, 2016-01-31:
Is there a way to have essentially the functionality of ``LT`` but also apply a modifier key the entire time the layer is active? 

In OSX, the app switcher is accessed with cmd+tab, and if you keep cmd depressed and hit tab repeatedly, you cycle through the apps. I'm hoping to make a layer with tab on one key and shift+tab on another, and have the layer activated by a key that also emulates holding cmd. 

I've searched for the answer and tried a bunch of things like combining the ``LT`` function with ``GUI_T`` or ``LGUI``, but haven't been able to accomplish it. Is it possible to do? Any guidance would be much appreciated, thanks!

jackhumbert, 2016-02-01:
You can accomplish this by using the macro code in #57 in combination with some layer switching!

```c
static uint16_t start;

const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt) 
{
    switch(id) {
        case 0:   
        if (record->event.pressed) {
            start = timer_read();
        	register_code(KC_LGUI);
        	layer_on(2);
        } else {
            unregister_code(KC_LGUI);
        	layer_off(2);
            if (timer_elapsed(start) < 150) {
                register_code(KC_ENT);
            }
        }
        break;
    }
}
```

jeffschenck, 2016-02-02:
Works like a charm — thank you so much!

jeffschenck, 2016-02-02:
Hm, sadly I may have spoken too soon. With the [macro in my code](https://github.com/jeffschenck/qmk_firmware/blob/dc116ac/keyboard/ergodox_ez/keymaps/jeffschenck/keymap.c#L217), based off the code above, when I hold the macro key (engaging the GUI modifier), tab around, then release the macro key (releasing the GUI modifier), I very frequently (like 80% of the time) end up in a state where a key is just on repeat. I haven't been able to detect a clear pattern to what gets repeated, but it seems to be inordinately often a "g" or a "9". Ever seen something like this happen?

jackhumbert, 2016-02-02:
That's pretty odd - I don't see any way that could happen through your keymap alone. You could try a `clear_keyboard();` in the `else` block (when the key is released) to see if that clears the repeating key.

jeffschenck, 2016-02-04:
Thanks, but it doesn't seem to help, unfortunately. Still getting odd characters being output or keys stuck on repeat, just holding then releasing the key assigned to the macro.

jackhumbert, 2016-02-04:
That's interesting - @ezuk, is this like anything you've seen before?

ezuk, 2016-02-04:
Nope... I doubt that it's hardware related tbh as it seems localized to this one macro?

jackhumbert, 2016-02-04:
Yeah, I agree - just odd that it's triggering keys that aren't at all related to the key(s) being pressed.

You're using qwerty, right?

jeffschenck, 2016-02-05:
Yup, it's essentially a qwerty layout. FWIW, I've put this idea aside for now so I can get the thing set up enough to start using regularly. I'd love to know the answer (and happy to keep trying things if there are more ideas), but it's not a major issue for me.

mecanogrh, 2016-02-05:
How do you mean the layer to be activated? One tap, double tap?
If the layer cmd key is also your default layer command it can't be a momentary layer switching action obviously. Is the layer cmd key also the default layer command key?
Is your goal to have cmd applied to all the layer so you don't have to depress cmd to send cmd+tab and cmd+shift+tab?

mecanogrh, 2016-02-05:
Why use timer here? Can't some use record->tap.count combined with record->tap.interrupted to achieve the same? Sorry not familiar with macros as I took the action_function() way.

mecanogrh, 2016-02-05:
Example (mod, key_tap and layer variables being all uint8_t passed to function on case switch statement) :
     ```
if (record->event.pressed) {
        if (record->tap.count > 0) {
            if (record->tap.interrupted) {
                record->tap.count = 0;  // ad hoc: cancel tap
                register_mods(MOD_BIT(mod));
                layer_on(layer);
            } else {
                register_code(key_tap);
            }
       } else {
         register_mods(MOD_BIT(mod));
         layer_on(layer);
       }
    } else {
        if (record->tap.count > 0) {
                unregister_code(key_tap);
        } else {
                unregister_mods(MOD_BIT(mod));
                layer_off(layer);
        }
    }

```

pattulus, 2016-02-21:
The [example](https://github.com/jackhumbert/qmk_firmware/issues/121#issuecomment-180321880) from @mecanogrh works for me and doesn't lead to random key repeat issues. Thanks.

mecanogrh, 2016-03-05:
check here [https://github.com/mecanogrh/qmk_firmware/blob/master/keyboard/planck/keymaps/mecano/keymap.c](url)

ezuk, 2016-07-24:
@pattulus is this still an issue?

closed 2017-06-27 by skullydazed