qmk/qmk_firmware #122: OSX System Keys

jeffschenck, 2016-01-31:
Is it possible to emulate the system keys you find on Apple keyboards for things like screen brightness up/down? I haven't been able to find any info on how to emulate those special keys. Thanks!

jackhumbert, 2016-02-01:
[Most aren't supported](https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#applemac-keyboard-fn), but [some are](https://www.reddit.com/r/olkb/comments/40k39p/how_to_assign_mac_brightness_keys/)!

```
KC_SLCK - Brightness up
KC_PAUS - Brightness down
```

wez, 2016-02-01:
Any idea if F14/F15 (also mentioned in that thread) will control the screen brightness in Windows 10, or what the keycodes to do that on Windows 10 might be?

jackhumbert, 2016-02-01:
I'm not familiar with Windows machines, but something like that may be possible with Autohotkey!

wez, 2016-02-01:
Was hoping for something native, but glad you pointed me to Autohotkey, thanks!

jeffschenck, 2016-02-02:
Thanks so much for the help! Unfortunately, I've tried using those keycodes and they don't seem to be working for me. Have others had success with KC_SLCK and KC_PAUS? (I'm on OSX 10.11.2.) Or is it something off with my code? (I have no experience with C... https://github.com/jeffschenck/qmk_firmware/blob/61a3ac0/keyboard/ergodox_ez/keymaps/jeffschenck/keymap.c#L185)

Any additional direction would be much appreciated! Otherwise, I can go back to mapping those keys to KC_F23 and KC_F24, then using BetterTouchTool to map those to brightness adjustments. It'd be great to have it cleanly contained in the keyboard, though. :)

wez, 2016-02-02:
F14/F15 worked to control the brightness both on my macbook and on my apple display when connected

jeffschenck, 2016-02-02:
@wez I definitely see those shortcuts set by default in System Preferences > Keyboard > Shortcuts > Display. However, they don't seem to work as advertised for me. Even when I set them to easier-to-press shortcuts (like Cmd+5) and hit the shortcut directly on my laptop's built-in keyboard, it doesn't trigger screen brightness. Perhaps it's just something off about my system...

romanzolotarev, 2016-03-19:
At first `KC_PAUS` and `KC_SLCK` didn't work for me neither, but after un-pluging and pluging the USB cable it started working. 

I have tried this with ErgoDox EZ on OS X 10.11.3 (15D21).

ezuk, 2016-03-20:
Thank you for following up here, @romanzolotarev! Too few people take the time to note when stuff works - super helpful. :)

romanzolotarev, 2016-03-20:
@ezuk Here is my layout https://github.com/jackhumbert/qmk_firmware/pull/205 :)
Note: I'm using `KC_F14` and `KC_F15` to control display brightness on OS X, but `KC_PAUS` and `KC_SLCK` also works.

ezuk, 2016-03-21:
Awesome! Saw the Readme, too :)

romanzolotarev, 2016-03-22:
Folks, here is a possible solution https://github.com/jackhumbert/qmk_firmware/pull/207. Could you please check?

closed 2016-06-18 by jackhumbert

moubry, 2018-10-21:
@jackhumbert’s solution worked for me, however it’s backwards (at least for me on Mojave) — `KC_SLCK` is bright down and `KC_PAUS` is brightness up.