qmk/qmk_firmware #128: Accessing system clock(s)

cbbrowne, 2016-02-06:
I'd like to grab values of one or another of the counters/clocks maintained on the CPU, to seed random processes.  (I implemented a wee random number generator so I have a key that emits randomly selected digits and another emitting randomly selected letters.)  Grabbing a clock value would make this *WAY* more random.

I see around 4 different clocks in the chip documentation; does anyone know how to get at their values from C code?

jackhumbert, 2016-02-06:
Yeah! I was meaning to reply on reddit, but this is much better :)

By default, I believe Timer 1 is always running, so you should be able to grab `TCNT1` to use for that. You could also set up 0, 3, or 4 as well! There's some nice info on the timers [here](http://www.avrbeginners.net/architecture/timers/timers.html) - the chip on the PCB/Teensy 2.0 is the Atmega32u4 if you need to find any specific info about it.

This is pretty awesome, though - let us know what happens!

mecanogrh, 2016-02-06:
Timer0 is already in use in tmk code maybe not the best idea to use it.
See timer_avr.h and timer.c in tmk_core/common/avr.

jackhumbert, 2016-02-06:
Sorry, that's the one I meant! I believe it's safe to read from `TCNT0` without affecting anything that's going on.

mecanogrh, 2016-02-06:
Something I don't get is that Timer0 is 8bits but timers in the code are 16 and 32bits in timer_read() functions, tried to implement a double tap/single tap on same key but timer_elapsed is not usable as is (only get 0 and 1 on key release). Began to read about atmega timers to get a finer one (so I can use whichever division I want without hacking the main code). My idea is to have 2 reference points I can #define in my keymap.c one for outputing a single tap if the release time is higher than for a double tap the second one being the interrupted on hold value (momentary press for layers or modifiers). Doing so will allow a single key to have single/double/hold combo. Well I guess :)

cbbrowne, 2016-02-07:
I'm grabbing the bottom 8 bits of TCNT1, and am finding it populated, for what it's worth.
I shut off my linear congruential RNG bit, and just used values from TCNT1, and was getting decently variable keystrokes, so it seems as though TCNT1 is being turned on.

With the combination of both RNGs (linear congruential, and XOR in bits from the counter), I'm certainly getting stuff looking random.

Here's some digits...  471074697342030825391029594704832

Here's some letters... xczqlozosvuslrmxcsbctuwtujsszbrwdzieoeegxssvqvecegszwaavkf

Looking good!  Thanks, very helpful!

closed 2016-02-07 by cbbrowne

jackhumbert, 2016-02-07:
That's great! Feel free to open a PR when you get things finalised :)

mecanogrh, 2016-02-09:
Was looking at your code cbbrowne and wondered where you put the timer init, looks like Timer1 is used by the backlight sleep code. But as you only need a read (thought you wanted to tailor the timer as well) you could have use Timer0 as well I guess. Sorry for the misunderstanding and noise guys.