qmk/qmk_firmware #143: ergodox_ez default keymap: hex not applying cmd when apostrophe is held

dudymas, 2016-02-15:
So I went ahead and nabbed the latest hex file from [keyboard/ergodox_ez/keymaps/default](-) since I saw the recent update to it.

From what I can tell, the hex flashes successfully, and teensy reboots just fine, but holding down apostrophe ( `'` ) yields no change. I just see apostrophe in my keymap.

I tried unplugging the ergodox to make sure, but indeed... I think the hexfile is bad. Can someone else confirm?

ezuk, 2016-02-17:
That's pretty weird -- this is the exact checksum I'm showing, and i just flashed this file and holding down ' (the right-hand pinky key in the third row) pops open the Start menu on Windows and sends Cmd on Mac. Would you mind sharing a pic or something showing exactly what key you're hitting?

dudymas, 2016-02-22:
Aha, okay, figured it out. I simply didn't understand the teensy programmer.
I was clicking the reboot button and not the reprogram button. :-/ I always assume if the arrow looks like a cycle, that's reboot, and if the arrow looks like it's pushing, then it's sending data. Durp.

Sorry about the confusion. And indeed, the apostrophe DOES map to the meta key now.
Closing.

closed 2016-02-22 by dudymas

ezuk, 2016-02-24:
Oh, nice! Thank you @dudymas!