qmk/qmk_firmware #144: Rapid press of two mousekeys makes mouse cursor stop.

tyetye, 2016-02-16:
Issue: If I press two mousekeys at the same time or almost the same time the mouse cursor stops moving.

Expected Behavior: I expect the mouse cursor to move on screen no matter how quickly I press multiple mousekeys.

Actual Behavior: The mouse cursor stops moving after moving a very short distance in the direction of the first mousekey press.

To Reproduce: Press a key assigned to KS_MS_U and then immediately press KS_MS_R. The mouse cursor should move diagonally to the upper right does instead it stops moving.

mecanogrh, 2016-03-03:
Could not reproduce here (planck), all is working fine and diagonals are working even if I (or so it seems) hit the two mouse keys at same time. Note that I'm using mousekeys on a different layer and have the layer key depressed as well all the time (momentary hold behaviour).

tyetye, 2016-03-04:
Interesting. I'm on an Ergodox EZ, but otherwise exactly as you described -- momentary hold for a new layer. I just did a new pull/compile and flashed the keyboard but the problem remains. Thanks for checking yours. I actually use the mouse keys a lot so it's kind of frustrating.

ezuk, 2016-03-05:
Hey @tyetye - just tested this on my ErgoDox EZ to see if it's maybe related to the keyboard. Here is what I noticed:

I can press a key mapped to "mouse up" and hold it down. Mouse starts moving up. I quickly add "mouse right" and get a diagonal. It's not hitting both at the same time -- it's a rolling motion, maybe 50msec separating both keydown events. To me, this is acceptable, but I understand if you need to press them both at exactly the same moment.

@jackhumbert - is there a reason for this to be different between the EZ and Planck? On your Planck with default QMK, can you hit both mousekeys at exactly the same time and get a diagonal?

tyetye, 2016-03-07:
Hi @ezuk. 
Yes, that's the behavior I am describing. 
It's not so much a matter of *needing* as much as it's frustrating when it doesn't work as expected. Imagine if the regular mouse worked as expected 90% of the time but 10% it just didn't move because one didn't initiate the move correctly.

ezuk, 2016-03-08:
Yup, understood. @jackhumbert any thoughts?

NoahAndrews, 2016-04-12:
It also works properly on my Planck. I'm curious to know what could cause this.

ezuk, 2016-04-12:
@jackhumbert - ping :) Any thoughts on this issue? Pretty interesting.

eltang, 2016-04-21:
@ezek What happens if you put the mouse keys on the right side of the keyboard, the side that's not handled by the I/O expander?

tyetye, 2016-04-21:
@eltang Interesting thought. My mouse keys are already on the right side with the Teensy. However, the layer modifier key is on the left hand side with the I/O expander.

eltang, 2016-04-21:
Uh-oh. I'll have to look into the code and see what might be causing the problem.

eltang, 2016-04-28:
@tyetye I'm planning to make some changes to the Ergodox EZ's `matrix.c` file. Let's see if you still have this problem after I do.

eltang, 2016-05-04:
@tyetye While solving #296, I noticed that the ErgoDox EZ has custom parameters for the mouse keys defined in `config.h`. What happens when you remove them?

algernon, 2016-05-11:
FWIW, I have experienced this problem too, on ErgoDox EZ, and it was very annoying, as I have diagonal mouse movement macros, which didn't work. For unrelated reasons, I moved the mouse keys to the right half (the layer key was already there), and now when I found this issue, and wanted to reproduce, I see it works now.

So it looks like @eltang got the right idea. Thanks!

tyetye, 2016-05-20:
@eltang @ezuk Thanks, eltang! It looks like that group is the culprit.
I set MOUSEKEY_DELAY=5 in place of the current value, MOUSEKEY_DELAY=100. This seems to have solved the issue. I haven't heavily tested to see if it interferes with non mouse related functions.

If I set MOUSEKEY_TIME_TO_MAX=25 and MOUSEKEY_DELAY=5, then the issue happens again. So, there is some sort of relationship there. I haven't explored it thoroughly.

ezuk, do you know if there is a reason these particular values were used?

ezuk, 2016-05-21:
Nope! Just inherited them, no particular reason for these. If you find values that work well, please feel free to submit a PR with them :)

eltang, 2016-05-21:
@tyetye @ezuk Those can be simply taken out. There are default values that will be used if no values are defined in `config.h`.

tyetye, 2016-05-24:
@eltang @ezuk I tried commenting out the values in the EZ's `config.h` so they use the QMK defaults. The mousekeys still work but become less smooth and less usable -- the movement of the mouse cursor is very jerky. This should probably be tested more thoroughly before completely removing the EZ's `config.h` values for everyone like in Fix #144.

eltang, 2016-05-24:
@tyetye That's why I didn't create a PR with those changes yet. As @ezuk asked, have you found a set of parameters that you like?

closed 2016-07-24 by ezuk