qmk/qmk_firmware #147: Support for KC_INT and some extra keys (KC_KP_COMMA) for an ABNT2 layout

potz, 2016-02-16:
Hello,

I am trying to create an Ergodox EZ firmware to be used with the Brazilian ABNT2 keyboard layout (the standard keyboard used in Brazil) on OS/X.

Now the ABNT2 is a strange little keyboard layout, not only it switches the keys around a bit (`;` becomes `Ç`, ``~` becomes `'"` etc), it also has two extra keys: one for the `/?` (scan code `0x73`, USB HID `0x87`) and other for the dot in the number pad (scan code `0x7e`, USB HID `0x85`). (The dot and comma are switched in the numpad because in brazil the comma is the decimal separator, I think).

Using the Massdrop configurator with Ben's firmware, I was able to produce these two keys with the codes `KEY_International1` for the `/?` and `KEYPAD_Comma` for the numpad dot. It works.

I can see from [The common keycodes in TMK core](https://github.com/jackhumbert/qmk_firmware/blob/master/tmk_core/common/keycode.h) that there are equivalent constants for these keys in TMK, namely `KC_INT1` and `KC_KP_COMMA`. However, I cannot get these to work on my Ergodox EZ.

I have done some diagnosing using the [Key Codes app](https://itunes.apple.com/us/app/key-codes/id414568915?mt=12) to check the codes actually being sent to the computer, and when using Ben's firmware the keys work and I receive the following information:

```
Key Down
  Characters:  /
  Unicode:     47 / 0x2f 
  Keys:        /
  Key Code:    94 / 0x5e
  Modifiers:   0 / 0x0

Key Down
  Characters:  .
  Unicode:     46 / 0x2e
  Keys:        .
  Key Code:    95 / 0x5f
  Modifiers:   256 / 0x100     # here I assume this is because I'm in the number layer
```

In my `keymap.c` file I used aliases to redefine the brazilian equivalent keys:  

```
#define BR_CCDL KC_SCLN      //  ;:  =>  çÇ
#define BR_SCLN KC_SLSH      //  /?  =>  ;:
#define BR_QUOT KC_GRV       //  `~  =>  '"
#define BR_TILD KC_QUOT      //  '"  =>  ~^
#define BR_AACT KC_LBRC      //  [{  =>  ´`
#define BR_LBRC KC_RBRC      //  ]}  =>  [{ 
#define BR_RBRC KC_BSLS      //  \|  =>  ]}
#define BR_BSLS KC_NUBS      //  #~  =>  \|   * (Non-US # and ~, sometimes § and ±)

#define BR_SLSH KC_INT1      //  /?  =>  scan code specific to abnt2 layout (slash/question mark)
#define BR_NDOT KC_KP_COMMA  //  .   =>  scan code specific to abnt2 layout (numpad dot)
```

But when compiling this and flashing the Ergodox EZ, keys using the `BR_SLSH` (`KC_INT1`) or `BR_NDOT` (`KC_KP_COMMA`) produce no effect, and the Key codes app doesn't show any event coming from the keyboard when the mapped keys are pressed.

[Here's my layout](https://github.com/potz/ergodox-ez-abnt2-layout/blob/master/keymap.c) for what is worth. The repository is at [potz/ergodox-ez-abnt2-layout](https://github.com/potz/ergodox-ez-abnt2-layout) and it has the script I have been using to compile it inside a [docker](https://github.com/docker/docker) container. It is a work in progress which I just started, I'm still trying out a couple of things but got stuck in this issue. I'd really prefer to use QMK rather than the massdrop configurator since it has a lot of extra features.

My questions are:

1. Are these extra keycodes supported in QMK at all? 

2. Am I doing something terribly stupid by creating these aliases? I just assumed this would work (it worked for the other switched keys), but my C skills are very very rusty. I also tried just sticking in the 0x87 and 0x85 codes with no success.

Thanks a lot,

potz

mecanogrh, 2016-02-17:
If you want /? use KC_SLSH, how can you have #define BR_SCLN KC_SLSH outputting a semicolon? That doesn't make sense. If you define BR_SCLN to KC_SLSH, BR_SCLN will output a slash.

potz, 2016-02-17:
@mecanogrh when using the Brazilian ABNT2 keyboard layout (the one you choose in your OS as an input source), the scan code usually used for the slash outputs a semicolon instead, among several other things that are switched. It's the same thing with the german keyboard layout that switches the `Y` for the `Z`, and the french layout that switches `Q` and `W` for `A` and `Z` (among other things).

This repo itself has [several mappings for alternate keyboard layouts](https://github.com/jackhumbert/qmk_firmware/tree/master/quantum/keymap_extras), I'm just doing doing exactly the same thing but for the specific mappings of the Brazilian ABNT2 layout. 

I hope it makes sense now. If it still doesn't may I suggest a little interesting experiment for you: change your OS keyboard layout temporarily to "Portuguese (Brazil ABNT2)" and type the slash key, see what gets written to your screen :)

You can ignore all the stuff about alternate layouts actually, the real issue is: how do I get the `KC_INT1` and `KC_KP_COMMA` to be actually sent by the hardware? They seem to be ignored at the moment.

mecanogrh, 2016-02-18:
Ok I understand now sorry, these input methods should be forbidden and all should be unicode coded in the input hardware :)
I tried KC_KP_COMMA on an US input method and it is indeed broken here as well (OSX). One  workaround if you need it straightaway is to give a shot to [Ukelele](http://scripts.sil.org/cms/scripts/page.php?site_id=nrsi&id=ukelele) if you are on OSX, it proved valuable for me to relocate special characters and dead keys (accents) before I got a planck.

ezuk, 2016-07-24:
@potz were you able to resolve this in the end? Or is this still a problem?

potz, 2016-07-24:
I have not solved this, no. It's been a while and I'm currently using a US-international variant but ideally I would like to get these keys working if I could only find a way.

shelaf, 2016-09-15:
@potz This problem seems to be similar to #312 .
If your problem is the same cause, would you like to try applying 6b8b332 patch or getting latest source code and rebuild your keymap.

potz, 2016-09-15:
YES!!! I can confirm that it works with the latest firmware!

Thank you very much, this is awesome! 😺 😸 😺  I'm gonna reactivate my little ABNT2 layout project now.

I updated my [docker image](https://hub.docker.com/r/potz/ergodox-ez-compiler/) to use the latest version of the firmware as well.

closed 2016-09-15 by potz