qmk/qmk_firmware #149: LT function does not switch back from higher layer after button release for certain keys

brokenalarms, 2016-02-16:
For LT(LAYERX, KEYX), it works as expected for KEYX keys like KC_ENT, KC_UP, even KC_Z (thus not just non-alphanumeric), but for certain keys (haven't isolated it exactly, but so far it appears to be the home row and KC_SPCE), holding the key just switches to layer LAYERX and gets stuck there. This means that unless you have a separate toggle switch for that particular layer, you're just stuck there forever....

Can try and fix myself too if you point me in the right direction (eg, where is the actual 'LT' function? I can't find it :P Coder but first foray into to C, apologies!)

Thanks!

brokenalarms, 2016-02-16:
My bad once reading instructions properly. Thanks :P

mecanogrh, 2016-02-16:
Post your keymap nobody can tell what is exactly happening without seeing it.

brokenalarms, 2016-02-17:
fixed, my issue. I didn't realize that held modifiers transforming to new layers need to be able to duplicate the original command on that new layer, either through KC_TRNS or repeating the original command. So my release events were being blocked by the assigned keys on the new layer.

Wouldn't have expected this behavior though...I'll rephrase and explain in new issue, thanks.

closed 2016-02-17 by brokenalarms