qmk/qmk_firmware #150: Remembering that key from lower layer has been pressed and held to go to higher layer

brokenalarms, 2016-02-17:
Use case:
I want to use thumb presses on each side to go temporarily toggle the media layer using LT(MDIA, KC_thmb):
- when pressing with my right thumb I'll be using the left half of the keyboard for mouse movement, with left thumb buttons for mouse clicks, and;
- when pressing with my left thumb I'll be using the right half of the keyboard for keyboard movement, with right thumb buttons for browser back/forward.

As I discovered, this doesn't work because the keys on the higher layer with different functions (though not in use because I'm using the other side of the board) take over and thus the release event from the lower layer is blocked.

The easiest fix is to make one layer with only left thumb and right keys, and another with right thumb and left keys. However this seems like it would get convoluted since I'd be needing to make a new layer for every different hand combination to find non-conflicting keys.

I'm imaging the complexity of the fix (since each key would have to remember the layer of its depressed state) is not worth the effort? Just wanted to confirm whether the fix above is the only way to handle this use case? Thanks!

ezuk, 2016-02-17:
Basically, the simplest thing would be to make two separate layers. I'm not sure why you'd have to make many others though -- you may get key duplication between those layers but that's okay I think?

brokenalarms, 2016-02-17:
Definitely the simplest option. Working on it. Thanks @ezuk .

closed 2016-02-17 by brokenalarms

ezuk, 2016-02-17:
sure thing!