qmk/qmk_firmware #156: LSFT (& friends) might affect the next key press

DidierLoiseau, 2016-02-21:
I am currently designing a layout in which some keys of the default layer need `LSFT`. However I noticed that when typing fast those keys, the shift state might affect the next key I press.

In fact, this occurs when I press the next key before I release the one with `LSFT`.

The problems comes from the implementation in `action.c`, where the shift key is actually maintained in the weak mods until the key is released:

```
        case ACT_LMODS:
        case ACT_RMODS:
            {
                uint8_t mods = (action.kind.id == ACT_LMODS) ?  action.key.mods :
                                                                action.key.mods<<4;
                if (event.pressed) {
                    if (mods) {
                        add_weak_mods(mods);
                        send_keyboard_report();
                    }
                    register_code(action.key.code);
                } else {
                    unregister_code(action.key.code);
                    if (mods) {
                        del_weak_mods(mods);
                        send_keyboard_report();
                    }
                }
            }
```

The advantage of this technique is that it allows the key repeat to work properly, with the above drawback.

Would it be possible to avoid this drawback? I'm thinking maybe clearing the weak mods when another key is pressed?

DidierLoiseau, 2016-02-21:
I have tested the clearing of weak mods on every key press and it seems to work properly:

```diff
diff --git a/tmk_core/common/action.c b/tmk_core/common/action.c
index 77ea39e..b8f3bf1 100644
--- a/tmk_core/common/action.c
+++ b/tmk_core/common/action.c
@@ -70,6 +70,10 @@
 #endif
     dprintln();
 
+    if (event.pressed) {
+        // clear the potential weak mods left by previously pressed keys
+        clear_weak_mods();
+    }
     switch (action.kind.id) {
         /* Key and Mods */
         case ACT_LMODS:
```

However it looks like the weak mods are used for `ACT_{L,R}MODS` but also for macros. It would probably make sense to only clear the mods applied by the former, so that it remains possible to set mods during a macro press, and release them on macro release.

It might be better to separate macro mods from the other weak mods then.

What do you think @jackhumbert ?

DidierLoiseau, 2016-02-25:
In fact I'm wondering why macros are using the weak mods instead of the normal mods. Does anyone have an idea? I have traced it down to cb3a547ebf but that does not really provide an explanation, except that it was intended.

BTW, if I changed the macros such that they use a separate bit field, would you merge such a PR @jackhumbert?

jackhumbert, 2016-02-25:
Sure! That would be great :)

I'm actually not sure why the weak mods are used in general, but I haven't a chance to dive too deep into things yet.

DidierLoiseau, 2016-02-25:
I think the idea is that weak mods are cleared when you change layers, to avoid issues with stuck mods (in case the new layer does not allow to properly release the mod), without clearing the normal mods (maintained by holding keys like `KC_RSFT`). This supposes that the target layer contains the mod keys at the same place though, but I guess it is most often the case.

The clearing is done in `clear_keyboard_but_mods()` (`action.c`), which is called on layer change – (`default_`)`layer_state_set()` in `action_layer.c`.

In fact, this could explain why weak mods were used for macros too, as a macro doesn't have the occasion to handle its key release when the target layer does not contain it.

commit 80eefb09 (2016-03-27): Merge pull request #188 from DidierLoiseau/issue-156

Fix #156: clear weak mods on every key press
commit b7a81f04 (2016-03-27): Fix #156: clear weak mods on every key press

- new macro_mods bit field for mods applied by macros
- weak_mods now only used for ACT_{L,R}MODS (i.e. LSFT, RSFT, LCTL etc.)
- clear the _weak_ mods on every key *pressed* such that LSFT etc.
  can no more interfere with the next key
closed 2016-03-27 by ezuk