qmk/qmk_firmware #173: Add special chars for us layout

GCorbel, 2016-03-04:
I'm trying to add a mode for french special chars in the colemak layout. In order to do that, I need to find a way to add special chars with the US layout. I think the simplest way is to use keycode, like 130 for eacute.

I tried to define a type like this : 

    #define FR_EACU	LALT(0x130)

I also tried with 0x0130 or 130 but it does not work.

Is there a way to do it?

GCorbel, 2016-03-07:
I seached again a solution and I didn't find anything. Is there a solution;

mecanogrh, 2016-03-07:
Get position of eacute on your OS input method, define it.
Example: on the OSX US Qwerty input method the Euro sign is produced by pressing Shift+Option+2 so I defined it with #define KC_EURO LSFT(LALT(KC_2)).
Now for the keycodes, look at `tmk_core/common/keycode.h`.

mecanogrh, 2016-03-07:
Note that keycodes handled in qmk are USB HID codes and not ASCII read the HID table here http://www.usb.org/developers/hidpage/Hut1_12v2.pdf

GCorbel, 2016-03-07:
@mecanogrh thanks for answer. I think some characters are not available with a single keystroke and have to use dead keys even with the us international keyboard or with the french layout like the character "ù".

I have to read the document you just gave.

Thanks for your help!

mecanogrh, 2016-03-07:
@GCorbel yes indeed, the thing to grasp here is that you are depending on the USB HID table to send keystrokes and that characters generation is handled by your input method. If you want to output *un e avec un accent aigu* you must either use the unicode capabilities of the firmware and have your OS input method set to unicode (but can be a complicated process) or use a macro to generate the dead key sequence then the e.

mecanogrh, 2016-03-07:
read the Macro examples on the front page here.

GCorbel, 2016-03-07:
I think marco is the answer. On linux, I can type Ctrl

mecanogrh, 2016-03-07:
You don't have to mix Macro and Unicode. You can use a Macro with the sequence you use to output the character.

GCorbel, 2016-03-07:
I think

    MACRO( T(LSHFT(CTL_T(U))), T(E), T(8), END  );

Will give è.

@mecanogrh you're right. It will produce the same result if I use normal keystroke. Typing unicode code allow to have chars like ⍅ or 玄 but I don't know it's useful.

Anyway, thanks for your help.

GCorbel, 2016-03-07:
It works with

    return MACRO(D(RSFT), D(LCTRL), T(U), U(RSFT), U(LCTRL), T(E), T(8), T(SPACE), END  );

I close this issue. @ezuk or @jackhumbert please, let me know if there is other solutions.

Thanks!

closed 2016-03-07 by GCorbel

ezuk, 2016-03-08:
@GCorbel not that I know.. @jackhumbert what do you think?

Way to go on solving this btw, thanks to @mecanogrh!

GCorbel, 2016-03-08:
For the moment, I did that : 

```
const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt)
{
  // MACRODOWN only works in this function
      switch(id) {
        case 0:
            if (record->event.pressed) {
              register_code(KC_RSFT);
            } else {
              unregister_code(KC_RSFT);
            }
            break;
        case 1:
            /* è */
            if (record->event.pressed) {
                return MACRO(D(RSFT), D(LCTRL), T(U), U(RSFT), U(LCTRL), T(E), T(8), T(SPACE), END  );
            }
        case 2:
            /* é */
            if (record->event.pressed) {
                return MACRO(D(RSFT), D(LCTRL), T(U), U(RSFT), U(LCTRL), T(E), T(9), T(SPACE), END  );
            }
        case 3:
            /* ê */
            if (record->event.pressed) {
                return MACRO(D(RSFT), D(LCTRL), T(U), U(RSFT), U(LCTRL), T(E), T(A), T(SPACE), END  );
            }
        case 4:
            /* ë */
            if (record->event.pressed) {
                return MACRO(D(RSFT), D(LCTRL), T(U), U(RSFT), U(LCTRL), T(E), T(B), T(SPACE), END  );
            }
        case 5:
            /* ù */
            if (record->event.pressed) {
                return MACRO(D(RSFT), D(LCTRL), T(U), U(RSFT), U(LCTRL), T(F), T(9), T(SPACE), END  );
            }
        case 6:
            /* ç */
            if (record->event.pressed) {
                return MACRO(D(RSFT), D(LCTRL), T(U), U(RSFT), U(LCTRL), T(E), T(7), T(SPACE), END  );
            }
        case 7:
            /* à */
            if (record->event.pressed) {
                return MACRO(D(RSFT), D(LCTRL), T(U), U(RSFT), U(LCTRL), T(E), T(0), T(SPACE), END  );
            }
      }
    return MACRO_NONE;
};
```

I can't do capital letter with special chars. Is there a function to change the macro if shift is pressed?

Thanks @ezuk I really like my Ergodox EZ!

ezuk, 2016-03-08:
Thank you @GCorbel!! :) That's so awesome to hear :)

mecanogrh, 2016-03-08:
@GCorbel you can use a new layer that triggers capitalized ù, é, à, etc. with macros at the same keys positions you used for lower cases.