qmk/qmk_firmware #174: Bootloader.hex file missing

NoahAndrews, 2016-03-04:
The new_project.sh script includes this line: `cp quantum/template/bootloader.hex keyboard/$KEYBOARD/bootloader.hex`. However, there is no bootloader.hex in that location. As far as I could find, there never has been either. We should either add a bootloader.hex file there or remove that line from the script. Which is the correct option? Is there any common scenario where the bootloader.hex file would be required?

jackhumbert, 2016-03-04:
Ah, yeah - I forgot to force add it, since this repo ignores .hex files. I originally added it in case people were doing ISP programming, but since the rest of the template assumes they're using a Teensy (on which you can't even modify the bootloader), I think it's safe to remove this `cp` line :)

NoahAndrews, 2016-03-04:
Pull request created.

commit 984c82a0 (2016-03-04): Merge pull request #175 from NoahAndrews/fix-#174

Fix #174
commit de1e2000 (2016-03-04): Fix #174
closed 2016-03-04 by jackhumbert