qmk/qmk_firmware #189: Matrix for 80 keys ergodox

krasnovu, 2016-03-09:
Hello! I can't understand. How i can modify files (config.h, keymap.c, ...) to use all 80 keys of my ergodox? Thanks!

krasnovu, 2016-03-10:
understood
i just added in ergodox_ez.h this define
```
#define KEYMAP80(                                               \
                                                                \
    /* left hand, spatial positions */                          \
    k00,k01,k02,k03,k04,k05,k06,                                \
    k10,k11,k12,k13,k14,k15,k16,                                \
    k20,k21,k22,k23,k24,k25,                                    \
    k30,k31,k32,k33,k34,k35,k36,                                \
    k40,k41,k42,k43,k44,                                        \
                            k55,k56,                            \
                        **k45,k46,**k54,                            \
                        k53,k52,k51,                            \
                                                                \
    /* right hand, spatial positions */                         \
        k07,k08,k09,k0A,k0B,k0C,k0D,                            \
        k17,k18,k19,k1A,k1B,k1C,k1D,                            \
            k28,k29,k2A,k2B,k2C,k2D,                            \
        k37,k38,k39,k3A,k3B,k3C,k3D,                            \
                k49,k4A,k4B,k4C,k4D,                            \
    k57,k58,                                                    \
    k59,**k47,k48,**                                                \
    k5C,k5B,k5A )                                               \
                                                                \
   /* matrix positions */                                       \
   {                                                            \
    { k00, k10, k20, k30, k40, KC_NO },   \
    { k01, k11, k21, k31, k41, k51 },   \
    { k02, k12, k22, k32, k42, k52 },   \
    { k03, k13, k23, k33, k43, k53 },   \
    { k04, k14, k24, k34, k44, k54 },   \
    { k05, k15, k25, k35, **k45,** k55 },   \
    { k06, k16, KC_NO,k36,**k46,** k56 },   \
                                                                 \
    { k07, k17, KC_NO,k37,**k47,** k57 },   \
    { k08, k18, k28, k38, **k48,** k58 },   \
    { k09, k19, k29, k39, k49, k59 },   \
    { k0A, k1A, k2A, k3A, k4A, k5A },   \
    { k0B, k1B, k2B, k3B, k4B, k5B },   \
    { k0C, k1C, k2C, k3C, k4C, k5C },   \
    { k0D, k1D, k2D, k3D, k4D, KC_NO }    \
   }
```
and use it instead of KEYMAP in my keymap.c
Thank you for reading this.

closed 2016-03-10 by krasnovu

ezuk, 2016-03-10:
Would be good to have this as its own keyboard if it makes it into the repo
(the EZ doesn't have 80 keys).