qmk/qmk_firmware #198: 1-setup-path-win.bat can corrupt your Windows system

IBNobody, 2016-03-16:
Just a heads-up, Jack. On Windows systems with existing and lengthy PATH environmental variables, SETX will corrupt your system.

```
Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Windows\system32>setx /M path "%PATH%;C:\MinGW\bin;C:\MinGW\msys\1.0\bin"

**WARNING: The data being saved is truncated to 1024 characters.**

SUCCESS: Specified value was saved.

C:\Windows\system32>
```

Since you mask the output to null, the user only sees "Success!", not the "oh by the way, your PATH statement is now corrupted" warning hint.

NoahAndrews, 2016-03-29:
Oh. Sorry for that. :anguished: Thanks for spotting this. (I was the one who wrote the offending script.) WHY doesn't Microsoft document this properly? I found an alternate solution and will submit a pull request.

IBNobody, 2016-03-29:
::Raises fist to the sky:: MICROSOFT!!!!

Seriously, though... Which fix are you going to use? SETX was nice because it sent out the notification that the env vars were changing. That is needed to tell Windows that you've updated the path.

NoahAndrews, 2016-03-29:
I think I'll modify the registry directly, and then use SETX to rewrite NUMBER_OF_PROCESSORS or something with itself. That way Windows will know that something changed.

IBNobody, 2016-03-29:
Maybe write to a scratch envvar, like SETX FOO=BAR, then delete it with SETX FOO=?

NoahAndrews, 2016-03-29:
only SET FOO= works, not SETX FOO=.

NoahAndrews, 2016-03-29:
https://github.com/jackhumbert/qmk_firmware/pull/223

closed 2016-03-29 by jackhumbert

IBNobody, 2016-03-30:
I rewrote the fix after testing it extensively. The problem is that Explorer.exe holds onto the old values no matter what. My new code restarts Explorer. (Take that, Windows!)

Hats off to @NoahAndrews for the initial idea.

NoahAndrews, 2016-03-30:
Why does it matter if explorer.exe holds onto the values?

IBNobody, 2016-03-30:
If you execute the batch file by double-clicking or if you execute it by opening a shift-right-click command prompt, Explorer retains the value of the environment variables. Even setx will not budge it. (Or at least setx setting a different key won't budge it.)

Thus, you can run into problems when you try to run your second batch file. The second batch file will fail on the first mingw-get call for systems that did not have the env vars set prior to the first batch file being run.

NoahAndrews, 2016-03-30:
Ah. :/

commit 26513b49 (2016-03-29): Merge pull request #223 from NoahAndrews/patch-3

Fix #198
commit 8f1ea16a (2016-03-29): Fix #198