qmk/qmk_firmware #202: Run code on key press

DavidBoyes, 2016-03-18:
I can't seem to find anywhere in a keyboard template where I could receive each key press - column and row - and run code based on that. Is this something that hasn't been implemented yet?

mecanogrh, 2016-03-19:
Yes you can have it, can't remember where precisely (maybe it's called in action.c) and the function may  not be declared in the header file but I've seen it.

DavidBoyes, 2016-03-19:
I have ended up hooking into process_action() in action.c. I couldn't find anything already existing though.

mecanogrh, 2016-03-19:
mmm you don't have to hack main code for this, in your keymap macro or in your keymap action_process() function if you are using it:
```c
keyevent_t event = record->event;
keypos_t key = event.key;
```
then access position with `key.col` and `key.row`

closed 2016-03-28 by DavidBoyes