qmk/qmk_firmware #204: Using LT(NAVI, KC_A) doesn't work

moonrumble, 2016-03-20:
as soon as I pressed A, it switches to NAVI layer permanently.

ezuk, 2016-03-20:
Interesting! Is your code available to look at? The whole layout, I mean.

moonrumble, 2016-03-21:
I have added my keymap.c in the repo.

https://github.com/moonrumble/qmk_layout.git

DidierLoiseau, 2016-03-21:
@moonrumble your problem is that the NAVI layer does not contain the same `LT` at the same place, so when you are on that layer, releasing the `A` key actually triggers a release of the `LEFT` key instead of the `LT`, and thus it cannot go back to the base layer. #182 is intended to fix this problem but in the meantime temporary layer keys need to have the same key on the destination layer.

BTW it would probably be easier to you to fork Jack's repo instead of creating a repo with only the `keymap.c`.

moonrumble, 2016-03-21:
I can test it tonight but I don't think otherwise other keys with LT modifier should not work as well. 

G and QUOTE keys work fine with LT modifier. 

Also on default layout, SYMB layer doesn’t use LT.

DidierLoiseau, 2016-03-21:
When you have `KC_TRNS` it is the same as repeating the key on the active layer below it. If you don't have any other layer active, it thus takes the value of the your BASE layer. That's why it works for all your other LT's.

moonrumble, 2016-03-22:
@DidierLoiseau Thansk, Now I understand the issue. If I use LT(NAVI, LEFT) in NAVI layer. should it work?

m1rado, 2016-04-05:
I think the best bet would be to use KC_TRNS on your NAVI layer.

closed 2016-04-06 by moonrumble