qmk/qmk_firmware #206: Virtual DIPs, Bootmagic change LALT with LGUI

krasnovu, 2016-03-21:
Hello!
I did not understand what I did but now LALT worked as LGUI. 
Where i can read step by step instructions with an example how reconfigure virtual dips ?

ezuk, 2016-03-21:
Hello! :)

Could you share a little more info on what your'e trying to accomplish?
What would you like to do with the virtual DIPs?

krasnovu, 2016-03-21:
I want reset my swaped keys LALT and LGUI

ezuk, 2016-03-21:
Is your layout anywhere on GitHub?

krasnovu, 2016-03-21:
```
[BASE] = KEYMAP80(
        // left hand
        KC_ESC, LSFT(KC_1),LSFT(KC_2),LSFT(KC_3),LSFT(KC_4),LSFT(KC_5),KC_NO,
        KC_TAB, KC_Q,      KC_W,      KC_E,      KC_R,      KC_T,      KC_NO,
        KC_NO,  KC_A,      KC_S,      KC_D,      KC_F,      KC_G,
        KC_ENT, KC_Z,      KC_X,      KC_C,      KC_V,      KC_B,      KC_NO,
        M(FLAT),KC_NO,     KC_LGUI,   MO(CURS),  MT(MOD_LSFT,KC_SPC),

                                                        KC_INS,  KC_PSCR,
                                  	           KC_BSPC, KC_DEL,  KC_NO,
                                               KC_LCTL, KC_LALT, KC_NO,
        // right hand
             KC_NO,LSFT(KC_6),LSFT(KC_7),         LSFT(KC_8),LSFT(KC_9),LSFT(KC_0),       KC_RBRC,
             KC_NO,KC_Y,      KC_U,               KC_I,      KC_O,      KC_P,             KC_LBRC,
                   KC_H,      KC_J,               KC_K,      KC_L,      LT(MDIA, KC_SCLN),KC_QUOT,
             KC_NO,KC_N,      KC_M,               KC_COMM,   KC_DOT,    KC_SLSH,          KC_ENT,
                              MT(MOD_RSFT,KC_SPC),MO(CURS),  KC_RGUI,   KC_BSLS,          M(FRUS),

             KC_APP, KC_BRK,
             KC_NO,  KC_ENT,  KC_BSPC,
             KC_NO,  KC_RALT, KC_RCTL
    ),
```
when i press key KC_LGUI its work like ALT and when i press key KC_LALT its work like GUI
but KC_RGUI and KC_RALT works fine

krasnovu, 2016-03-22:
Morning is smarter than evening!
I read bootmagic readme again and resolve my issue.
I am hold Backspace key with Space key and plug in keyboard cable.
My stored configuration cleared in eeprom.

closed 2016-03-22 by krasnovu

ezuk, 2016-03-22:
Nice! Glad to hear you resolved it.