qmk/qmk_firmware #208: OSX Gui Key's Not Working

braidn, 2016-03-23:
Just pulled the most recent version of QMK and decided to tweak my planck layout. However on reboot, the bottom row of the following keymap was acting off:

https://gist.github.com/d2b3cd4986b61178c2af

LGUI fires as if it's an LALT, LALT doesn't fire at all, RGUI doesn't fire at all, and RALT fires fine. All of these fire as 'FlagsChanged' in event viewer over the more traditional keyup/keydown I have never seen this before but it is definitely odd. Any help would be amazing. Thanks again

braidn, 2016-03-23:
Ok, so apparently this is due to a Boot Magic mode that the keyboard is in? Holding space + backspace on the keyboard as you plug it in seems to fix everything. Closing but updating for history's sake and if someone is searching.

closed 2016-03-23 by braidn

tuananh, 2018-11-21:
@braidn a thousand upvotes. I love you!!!!