qmk/qmk_firmware #212: Sticky/Dead Key/Oneshot Modifier for Layer Switching?

chrischambers, 2016-03-26:
Is it possible to map a a sticky/deadkey for layer switching? E.g. hitting capslock followed by the "a" key results in whatever's mapped to the "a" key in layer 1? I see that there's a  [oneshot modifier](https://github.com/jackhumbert/qmk_firmware/blob/master/tmk_core/doc/keymap.md#43-oneshot-modifier) action, but I've not managed to make it work for layer switching.

As another twist - is it possible to have the capslock key, for example, act as a sticky key when tapped and a momentary layer shift when held? Similar to the ergodex ez default keymap 'temp coder layer', but with a sticky key mode instead of a toggle when tapped?

chrischambers, 2016-03-27:
An update: I see that [ahtn](https://github.com/ahtn) put in some work [here](https://github.com/tmk/tmk_keyboard/pull/255) to make this possible in the original repository, which has yet to be merged. I created a [patch](https://gist.github.com/chrischambers/748467e46db72d6c0dea) which applies cleanly but didn't have the desired result (it rendered the left half of my ergodox unresponsive).

Delapouite, 2016-03-29:
Yes. Dead keys are also particulary useful to deal with accentuated letters in many Europeans languages.

Townk, 2016-05-05:
Just to update this, I just created a pull request that has no conflicts with the master branch here: https://github.com/jackhumbert/qmk_firmware/pull/308

Townk, 2016-05-09:
I think this can be closed since #308 was merged.

closed 2016-05-12 by ezuk