qmk/qmk_firmware #221: LGUI(KC_LSFT) does not work

sboesebeck, 2016-03-29:
Hi,

when defining a key to be LGUI(KC_LSFT) it remains pressed until I hit another key. Then LGUI is switched off and only LSFT is pressed... so instead of LGUI_LSFT_N you type LSFT_N

when swapping LGUI and LSFT the behavior is similar, only LSFT is dropped out and only LGUI being pressed...

Hope this makes sense of some kind ;-)

sboesebeck, 2016-03-29:
I added a macro doing the same, that worked fine

ezuk, 2016-04-01:
@jackhumbert any reason why this wouldn't work? I think the weak mods PR fixed it?

DidierLoiseau, 2016-04-01:
I think #188 broke this usage, which I had never thought of. For me the usage of `LSFT`, `LGUI` etc. is designed to only affect a single key press, not the next ones.

It could make sense to apply a different behavior when those are applied to `KC_LSFT` etc., by using the normal mods instead of the weak mods for example.

Note that the PR was especially designed to allow this usage through macros though.

closed 2016-04-06 by ezuk

sboesebeck, 2016-04-06:
Hi - I implemented the behaviour using a macro, thanks for clearification!

DidierLoiseau, 2016-04-06:
@sboesebeck As #239 has now been merged, you shouldn't need the macro any more :-)

PR #239 (2016-04-06): Fix issue #221: LGUI(KC_LSFT) does not work
Fixes #221 by registering `LGUI`, `LSFT` etc. on mod keys as normal mods instead of weak mods:
 - they won't be cleared when pressing another key (#188)
 - they won't be cleared by layer switching
 - `LSFT(KC_LGUI)` will now have the same behavior as `LGUI(KC_LSFT)`