qmk/qmk_firmware #237: Waking up Mac from sleep

janv, 2016-04-05:
I'm using an Ergodox EZ with the QMK firmware.
The keyboard does not seem to be able to wake my Mac from sleep.
Is there a way to make that work?

ezuk, 2016-04-06:
Sure! See: https://github.com/tmk/tmk_keyboard/wiki/FAQ

Specifically:

Wakeup from sleep doesn't work

In Windows check Allow this device to wake the computer setting in Power Management property tab of Device Manager. Also check BIOS setting.

Pressing any key during sleep should wake host.

ezuk, 2016-04-06:
Did you try `KC_WAKE` and `KC_PWR`?

jackhumbert, 2016-04-06:
You might need to hold down the key as well - it doesn't help that OSX's sleep sucks :/

janv, 2016-04-16:
Hm, could't get it to work.
But it's not that big a deal I guess, I can use my mouse to wake up the computer

closed 2016-04-16 by janv

jtbandes, 2016-06-01:
I'm experiencing this as well. I'll try KC_WAKE/PWR soon.

But why is this necessary? Other keyboards/mice I've used (usually Apple ones) always seem to wake the machine when something is pressed, not just the power key.

cleung2010, 2016-06-28:
I am experiencing the opposite issue on the Planck, where the keyboard stops working after a wakeup. The backlights are still on, so I know that it's not a power issue. 

This started occurring after I updated the firmware to the commit `2cf26915e3ed737ae83cf7b54f7e84fc2f0d3b40` (Jun. 16). Before that, I was running on the build from `3c683aa9f23288ddada760fb49abcc5fa7324f5e` (Feb. 7) with no problems.

The custom keymap that I flashed is based off `default` with some keys swapped, but no major changes.

jackhumbert, 2016-06-28:
@cleung2010 what OS are you on? I've had something similar recently, where the matrix seems to be acting up. I'm wondering if it's related to #444.

cleung2010, 2016-06-28:
This is on OSX El Capitan.

janv, 2017-03-06:
I updated my firmware last week and at least pressing `SPACE` now seems to wake up my mac

bennypowers, 2017-05-10:
I have this same issue. Shouldn't have to issue a special `WAKE` command - any keypress should wake from sleep.