qmk/qmk_firmware #240: KC_AT with right shift

moonrumble, 2016-04-06:
Instead of numbers, I have used their shift counterpart. 

While I am still getting used to my layout. I often press right shift key and press KC_AT key. 

If I press LSHIT and KC_AT, it prints KC_AT but if I press RSHIFT and KC_AT it changes the layer to next layer.

moonrumble, 2016-04-06:
https://github.com/moonrumble/qmk_layout/blob/master/keymap.c

ezuk, 2016-04-06:
Is there a particular reason why you're even using RSHIFT? What happens if you define both Shift keys on your keyboard as LSHIFT? Do you get any ill effects?

jackhumbert, 2016-04-06:
There's some fancy stuff with KC_AT going on behind the scenes - my guess is that right shift gets combined with KC_AT (which is really LSFT and KC_2) to switch to layer 2. I think that's a magic key combo to switch to layer *x*, but I could be mistaken, and it could have to do with the bits passed around by both shifts.

ezuk, 2016-04-06:
Ah right - so RSHFT + KC_AT really means RSHFT + LSHFT + 2. So having both shift keys as LSHFT should resolve.

jackhumbert, 2016-04-21:
You can also change the command combo [here](https://github.com/jackhumbert/qmk_firmware/blob/master/keyboard/ergodox_ez/config.h#L66) - let us know if it works!

closed 2016-05-02 by jackhumbert