qmk/qmk_firmware #245: Add more detailed instructions for installation of MinGW in BUILD GUIDE

VoodaGod, 2016-04-10:
I followed the instructions for setting up the build environment for windows, and in step 3, it reads "Install MinGW. During installation, uncheck the option to install a graphical user interface. DO NOT change the default installation folder. The scripts depend on the default location."
I did exactly this and the scripts did not function properly, even though I had installed MinGW in the default location.
However once I did not uncheck the option to install a GUI, I was able to see that MinGW had actually not been installed yet, and I had to manually select apropriate packages for installation.
I installed the packages I considered relevant (please list the required ones explicitly) and tada! the scripts work.

In conclusion I could have saved myself quite a bit of time if the build guide had not told me to uncheck the "install GUI" option, and I'm sure there are many more noobs like me that would appreciate the few extra steps in this guide to tell us what exactly needs to be installed for the scripts to work.

NoahAndrews, 2016-04-10:
The instructions should work as-is. All that the second script needs MinGW-wise is the base MinGW install, which it then uses to install the required packages. Since the GUI tool isn't necessary thanks to the scripts, I figured people wouldn't want it cluttering up their programs list.

> the scripts did not function properly

Please be more specific. What was the output? How did you know they went wrong?

> However once I did not uncheck the option to install a GUI, I was able to see that MinGW had actually not been installed yet

If you're running the GUI tool, MinGW has already been installed.

> I installed the packages I considered relevant (please list the required ones explicitly) and tada! the scripts work.

Which packages did you install manually here? Again, any required packages should have been installed by the second script. I'd much rather just make sure the script works as expected rather than make the end user install the packages themselves. 

Thanks for reporting your issue!

VoodaGod, 2016-04-10:
Ok, i'll try to recreate the issue on another pc, i'll get back to you when i have more info

VoodaGod, 2016-04-11:
i followed the instructions on another machine and had the same issue:

run 1-path-setup-win.bat: in cmd (Path not found: "C:\MinGW\msys\1.0\bin")
run 2-setup-environment-win.bat: 2 popup windows:

	cp.exe - System Error(The program can't start because msys-intl-8.dll is missing from your computer. Try reinstalling the program to fix this problem) -> ok
	rm.exe - System Error (The program can't start because msys-intl-8.dll is missing from your computer. Try reinstalling the program to fix this problem) -> ok
	
	cmd log: 
	------------------------------------------
	Installing wget and unzip
	------------------------------------------
	'mingw-get' is not recognized as an internal or external command,
	operable program or batch file.
	A subdirectory or file temp already exists.
	------------------------------------------
	Installing dfu-programmer.
	------------------------------------------
	'wget' is not recognized as an internal or external command,
	operable program or batch file.
	'unzip' is not recognized as an internal or external command,
	operable program or batch file.
	The system cannot find the file specified.
	------------------------------------------
	Downloading driver
	------------------------------------------
	'wget' is not recognized as an internal or external command,
	operable program or batch file.
	'unzip' is not recognized as an internal or external command,
	operable program or batch file.
	------------------------------------------
	Installing driver. Accept prompt.
	------------------------------------------
	Microsoft PnP Utility

	Adding the driver package failed : Invalid INF passed as parameter.

	Total attempted:              0
	Number successfully imported: 0

	------------------------------------------
	Finished!
	Press any key to continue . . .

run make in cmd as admin for example in keyboard/planck
	cmd log:

	C:\Users\Test\Documents\GitRepos\qmk_firmware\keyboard\planck>make
	The system cannot find the path specified.
	The system cannot find the path specified.
	ECHO is off.
	-------- begin --------
	avr-gcc (GCC) 4.8.2 20131010 (prerelease)
	Copyright (C) 2013 Free Software Foundation, Inc.
	This is free software; see the source for copying conditions.  There is NO
	warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

	-f was unexpected at this time.
	../../tmk_core/rules.mk:395: recipe for target 'sizebefore' failed
	make: *** [sizebefore] Error 255

NoahAndrews, 2016-04-11:
Thank you so much for posting this. Pull request #247 should fix this.

closed 2016-04-11 by VoodaGod