qmk/qmk_firmware #250: Macros with non US keycodes

sboesebeck, 2016-04-12:
Hi,

the MACRO function takes the keycode definition without the KC_ prefix, if I see that right. This does cause some problems, if you want to create a macro on a different international layout. You always need to think "When I type this on my layout, what would that be on an US one?"

Is there a way to have the proper prefixes available for macros as well. As I see it, there are already a lot of language specific definitions?

This would macro creation a lot easier - at least for the simple macros.

sboesebeck, 2016-04-20:
found it myself. Using UP, DOWN and TYPE instead of U,D,T works...

closed 2016-04-20 by sboesebeck

ezuk, 2016-04-21:
@sboesebeck what can I say? :) You rock. I was going to pester @jackhumbert about this but you beat me to it :)