qmk/qmk_firmware #255: Per keymap makefile.mk and changing *_ENABLE for the better

jackhumbert, 2016-04-15:
[Here's the example](https://github.com/jackhumbert/qmk_firmware/blob/new_defaults/keyboard/planck/Makefile) - you can throw in a makefile.mk into your keymap folder (keymaps/*keymap*/makefile.mk) for it to be included. This only works when the keymap is a folder (and not a file - something that is still compatible with the Makefile). The default options needed to be moved up in the keyboard Makefile to make this effective. The default example there uses the AUDIO_ENABLE that you can change.

*_ENABLE also looks for "yes" instead of just being defined, so you can override things with "no" instead of having to undefine them from "yes". Previously, a lot of code looked like this:

    ifdef *_ENABLE

and now it looks like:

    ifeq ($(*_ENABLE), yes)

I think I caught all of these (enough so it compiles), but some more thorough checking is probably necessary.

@IBNobody and @skullydazed talked about this [here](https://www.reddit.com/r/olkb/comments/4eskto/best_approach_for_qmk_git_branches_custom/), and I welcome your comments!

jackhumbert, 2016-04-15:
This is a little different than the implementation in #151 - I'm not sure which way is more ideal, or the real differences between `=` and `:=`.

IBNobody, 2016-04-15:
By the way... Is there a reason you stripped out some of the extra make recipes / flags in your general makefile? 

```
debug-on: EXTRAFLAGS += -DDEBUG -DDEBUG_ACTION
debug-on: all

debug-off: EXTRAFLAGS += -DNO_DEBUG -DNO_PRINT
debug-off: OPT_DEFS := $(filter-out -DCONSOLE_ENABLE,$(OPT_DEFS))
debug-off: all
```

I noticed that DEBUG_ACTION was a key being used by action_layer. Maybe that should be a define?

IBNobody, 2016-04-15:
Oh... Also... Can you make it Y/N? Or T/F? That way, we won't have to worry about spaces and keeping the comments aligned.

jackhumbert, 2016-04-15:
The only place I could find that stuff was [here](https://github.com/tmk/tmk_keyboard/blob/81c3c1082a59c70236da56fbda505f1432cf27c6/keyboard/hhkb/Makefile) - I'm guessing it wasn't in the GH60 project when it got copied :) 

I thought about changing it up, but keeping it as yes/no allows it to be a little bit backwards compatible for the older/less maintained projects, and things downstream.

NoahAndrews, 2016-04-15:
That's great that you can finally disable features with "no"! Looks like you're still using the old style in the keymap file though.

jackhumbert, 2016-04-15:
The new style actually only applies to the makefiles - in .c files, you can (and should, I guess) use the ifdef :)

NoahAndrews, 2016-04-15:
Of course, #ifeq isn't a C preprocessor directive. So where do those get defined for the C files? Does it have something to do with OPT_DEFS in the makefiles? 

I apologize for any noobish questions. C & C++ may have been my first programming languages, but I never got good at them. I also don't have any experience with makefiles outside of working with QMK. I plan to work to understand QMK much more thoroughly over the summer.

jackhumbert, 2016-04-15:
Yeap! [Here's the majority of them](https://github.com/jackhumbert/qmk_firmware/blob/new_defaults/tmk_core/common.mk) - it'd probably make more sense if they had different names :)

NoahAndrews, 2016-04-15:
Yeah, that's what I saw that lead me to that conclusion. But how/where does OPT_DEFS translate to defining things in the C preprocessor?

jackhumbert, 2016-04-15:
My understanding of it is that -D*VARIABLE* (which gets passed directly to the compiler) just defines *VARIABLE* on the C side of things.

IBNobody, 2016-04-15:
The -d flag converts them. That's how I stumbled on DEBUG_ACTION

NoahAndrews, 2016-04-15:
Gotcha, thanks for the explanation

IBNobody, 2016-04-15:
So... The keymap makefile overrides any settings?

jackhumbert, 2016-04-15:
It should overwrite the settings that come before it - you can see where the file gets included in the Makefile.

I seem to have broken the backlight somehow, though.

IBNobody, 2016-04-15:
This is going to be so nice when we get it working...

IBNobody, 2016-04-15:
Move `CONFIG_H = ` up so we can override it.

Also move this up.

```
ifeq ($(BACKLIGHT_ENABLE), yes)
	SRC += backlight.c
endif
```

On my end, all my printf's are broken.

jackhumbert, 2016-04-15:
Just pushed 43a4ffc (how do you reference commits?) - we might have broken the bootmagic include somehow.

IBNobody, 2016-04-15:
Oh... Wait... Moving the backlight is going to cause problems because I want to turn the backlight off, and moving the `SRC += backlight.c` line set higher will mean that the backlight is enabled for me.

jackhumbert, 2016-04-15:
Yeah, apparently all of the settings are too high - I left out the backlight setting in the makefile.mk, and it's not getting turnt on by the defaults set by Makefile.

jackhumbert, 2016-04-15:
It's the spaces!! I understand the need for `($(strip $(BACKLIGHT_ENABLE)), yes)` now :)

jackhumbert, 2016-04-15:
Done and did in #256!

closed 2016-04-15 by jackhumbert