qmk/qmk_firmware #268: Planck build error when customizing default keymap

ericlathrop, 2016-04-20:
This works fine:
```
cd keyboard/planck
make
```
This explodes:
```
cp keymaps/default/keymap.c keymaps/eric.c
make KEYMAP=eric
```
with these errors:
```
-------- begin --------
avr-gcc (GCC) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
  24556     436     262   25254    62a6 planck.elf

mkdir -p obj_planck/keymaps
Compiling C: keymaps/eric2.c
avr-gcc -c -mmcu=atmega32u4 -gdwarf-2 -DF_CPU=16000000UL -DINTERRUPT_CONTROL_ENDPOINT -DBOOTLOADER_SIZE=4096 -DF_USB=16000000UL -DARCH=ARCH_AVR8 -DUSB_DEVICE_ONLY -DUSE_FLASH_DESCRIPTORS -DUSE_STATIC_OPTIONS="(U
SB_DEVICE_OPT_FULLSPEED | USB_OPT_REG_ENABLED | USB_OPT_AUTO_PLL)" -DFIXED_CONTROL_ENDPOINT_SIZE=8  -DFIXED_NUM_CONFIGURATIONS=1 -DPROTOCOL_LUFA -DMOUSEKEY_ENABLE -DMOUSE_ENABLE -DEXTRAKEY_ENABLE -DNO_PRINT -DNO
_DEBUG -DCOMMAND_ENABLE -DBACKLIGHT_ENABLE -DVERSION=f489ae1-dirty -Os -funsigned-char -funsigned-bitfields -ffunction-sections -fdata-sections -fno-inline-small-functions -fpack-struct -fshort-enums -fno-strict
-aliasing -Wall -Wstrict-prototypes -Wa,-adhlns=obj_planck/keymaps/eric2.lst -I. -I../.. -I../../tmk_core -I../../quantum -I../../quantum/keymap_extras -I../../tmk_core/protocol/lufa -I../../tmk_core/protocol/lu
fa/LUFA-git -I../../tmk_core/common -std=gnu99 -include config.h -MMD -MP -MF .dep/obj_planck_keymaps_eric2.o.d  keymaps/eric2.c -o obj_planck/keymaps/eric2.o 
keymaps/eric2.c:180:31: error: ‘AUD_ON’ undeclared here (not in a function)
   {_______, _______, _______, AUD_ON,  AUD_OFF, AG_NORM, AG_SWAP, QWERTY,  COLEMAK, DVORAK,  PLOVER,  _______},
                               ^

keymaps/eric2.c:180:40: error: ‘AUD_OFF’ undeclared here (not in a function)
   {_______, _______, _______, AUD_ON,  AUD_OFF, AG_NORM, AG_SWAP, QWERTY,  COLEMAK, DVORAK,  PLOVER,  _______},
                                        ^
keymaps/eric2.c: In function ‘process_action_user’:
keymaps/eric2.c:357:9: warning: implicit declaration of function ‘play_note’ [-Wimplicit-function-declaration]
         play_note(((double)220.0)*pow(2.0, -4.0)*pow(2.0,(starting_note + SCALE[record->event.key.col + offset])/12.0+(MATRIX_ROWS - record->event.key.row)), 0xF);
         ^
keymaps/eric2.c:357:9: warning: implicit declaration of function ‘pow’ [-Wimplicit-function-declaration]
keymaps/eric2.c:357:35: warning: incompatible implicit declaration of built-in function ‘pow’ [enabled by default]
         play_note(((double)220.0)*pow(2.0, -4.0)*pow(2.0,(starting_note + SCALE[record->event.key.col + offset])/12.0+(MATRIX_ROWS - record->event.key.row)), 0xF);
                                   ^
keymaps/eric2.c:357:75: error: ‘SCALE’ undeclared (first use in this function)
         play_note(((double)220.0)*pow(2.0, -4.0)*pow(2.0,(starting_note + SCALE[record->event.key.col + offset])/12.0+(MATRIX_ROWS - record->event.key.row)), 0xF);
                                                                           ^
keymaps/eric2.c:357:75: note: each undeclared identifier is reported only once for each function it appears in
keymaps/eric2.c:357:113: error: invalid operands to binary / (have ‘const uint16_t (*)[12]’ and ‘double’)
         play_note(((double)220.0)*pow(2.0, -4.0)*pow(2.0,(starting_note + SCALE[record->event.key.col + offset])/12.0+(MATRIX_ROWS - record->event.key.row)), 0xF);
                                                                                                                 ^
keymaps/eric2.c:357:9: error: incompatible type for argument 2 of ‘pow’
         play_note(((double)220.0)*pow(2.0, -4.0)*pow(2.0,(starting_note + SCALE[record->event.key.col + offset])/12.0+(MATRIX_ROWS - record->event.key.row)), 0xF);
         ^
keymaps/eric2.c:357:9: note: expected ‘double’ but argument is of type ‘const uint16_t (*)[4][12]’
keymaps/eric2.c:357:49: error: invalid operands to binary * (have ‘double’ and ‘const uint16_t (*)[4][12]’)
         play_note(((double)220.0)*pow(2.0, -4.0)*pow(2.0,(starting_note + SCALE[record->event.key.col + offset])/12.0+(MATRIX_ROWS - record->event.key.row)), 0xF);
                                                 ^
keymaps/eric2.c:359:9: warning: implicit declaration of function ‘stop_note’ [-Wimplicit-function-declaration]
         stop_note(((double)220.0)*pow(2.0, -4.0)*pow(2.0,(starting_note + SCALE[record->event.key.col + offset])/12.0+(MATRIX_ROWS - record->event.key.row)));
         ^
keymaps/eric2.c:359:35: warning: incompatible implicit declaration of built-in function ‘pow’ [enabled by default]
         stop_note(((double)220.0)*pow(2.0, -4.0)*pow(2.0,(starting_note + SCALE[record->event.key.col + offset])/12.0+(MATRIX_ROWS - record->event.key.row)));
                                   ^
keymaps/eric2.c:359:113: error: invalid operands to binary / (have ‘const uint16_t (*)[12]’ and ‘double’)
         stop_note(((double)220.0)*pow(2.0, -4.0)*pow(2.0,(starting_note + SCALE[record->event.key.col + offset])/12.0+(MATRIX_ROWS - record->event.key.row)));
                                                                                                                 ^
keymaps/eric2.c:359:9: error: incompatible type for argument 2 of ‘pow’
         stop_note(((double)220.0)*pow(2.0, -4.0)*pow(2.0,(starting_note + SCALE[record->event.key.col + offset])/12.0+(MATRIX_ROWS - record->event.key.row)));
         ^
keymaps/eric2.c:359:9: note: expected ‘double’ but argument is of type ‘const uint16_t (*)[4][12]’
keymaps/eric2.c:359:49: error: invalid operands to binary * (have ‘double’ and ‘const uint16_t (*)[4][12]’)
         stop_note(((double)220.0)*pow(2.0, -4.0)*pow(2.0,(starting_note + SCALE[record->event.key.col + offset])/12.0+(MATRIX_ROWS - record->event.key.row)));
                                                 ^
keymaps/eric2.c: At top level:
keymaps/eric2.c:371:6: warning: function declaration isn’t a prototype [-Wstrict-prototypes]
 void play_goodbye_tone()
      ^
keymaps/eric2.c: In function ‘play_goodbye_tone’:
keymaps/eric2.c:373:3: warning: implicit declaration of function ‘PLAY_NOTE_ARRAY’ [-Wimplicit-function-declaration]
   PLAY_NOTE_ARRAY(goodbye, false, 0);
   ^
keymaps/eric2.c:373:19: error: ‘goodbye’ undeclared (first use in this function)
   PLAY_NOTE_ARRAY(goodbye, false, 0);
                   ^
keymaps/eric2.c:374:3: warning: implicit declaration of function ‘_delay_ms’ [-Wimplicit-function-declaration]
   _delay_ms(150);
   ^
make: *** [obj_planck/keymaps/eric2.o] Error 1
```
Am I doing something wrong, or is master broken?

jackhumbert, 2016-04-20:
It looks like you might be missing a lot of the definitions at the top of the default file, oddly enough. It'd probably work best to copy the folder `default` folder instead of just the keymap file. There's a lot of audio stuff in the default keymap right now that hasn't been properly `#ifdef`-ed.

It's very possible that other things are broke right now though :) I'm finalising the default layouts for the Massdrop!

ericlathrop, 2016-04-20:
That worked, thanks!

closed 2016-04-20 by ericlathrop