qmk/qmk_firmware #269: [Enhancement] (Possibly existing) Use normal key as a hold button.

BeatVids, 2016-04-21:
Currently, we can use a button for tapping, while having another function for holding.
i.e. tap z, hold ctrl:   CTL_T(KC_Z)

I was wondering if this was possible not just with modifiers, but with normal keys, i.e.:
**Tap** a certain button, use it as VolDwn.
**Hold** that same button, and use it as Mute.

If this isn't currently possible, it would be an awesome feature, here are some examples I think would be practical and handy:
```
**Tap** KC_WWW_HOME     **Hold** KC_WWW_SEARCH
**Tap** KC_MEDIA_PLAY   **Hold** KC_MEDIA_STOP
**Tap** KC_VOLDOWN      **Hold** KC_MUTE
**Tap** KC_ESCAPE       **Hold** Alt+F4
**Tap** KC_n            **Hold** KC_ñ
```

For these normal keys, it may be best to not actually hold the key, but to activate it only once as soon as it has been held for the specified amount of time.

ezuk, 2016-04-21:
You know, I just thought this might be possible to do as a macro, with the [timer functionality](https://github.com/jackhumbert/qmk_firmware#timer-functionality). @jackhumbert is there a simpler way?

jackhumbert, 2016-04-21:
I think the timer is the easiest way! Like you mentioned, the hold keypress won't be sent until it's had time to check to see if the key's been released for the tap keypress. 

Something like `TH(LCTL(V), LCTRL(C))` would be cool to implement - we should be able to compile the scancodes and modifiers together, but possibly not with the 16bit keycodes we have currently. Just more reasons to expand to 32bit ;)

mecanogrh, 2016-04-21:
You don't need to care to handle the timer yourself here (unless you want per key timeout values). QMK/TMK already has a way of doing this. All you need is:
```c
if (event.pressed) {
  if (record->tap.count>0) {
    // this is a tap
  } else {
    // this is an hold
  }
} else {
  // this is the key release
  if (record->tap.count>0) {
    record->tap.count = 0; // needed to reset key tap count on release
  }
}
```

the threshold in ms is defined with:
```c
#define TAPPING_TERM 200
```

put a macro on the key you want and you are good to go.
I don't use this through macros (I use action_function() instead) but there is no reason is shouldn't work there I guess.

BeatVids, 2016-04-21:
Hi guys, I'm so sorry to do this, but would you mind fixing my code? I only have about a day experience with C! I have set a button to LCTL(KC_C), now I just don't know how to detect it or set the action for it.
@jackhumbert , that  TH() format looks super promising!

```
static uint16_t key_timer;

const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt)
{
        case LCTL(KC_C):
        key_timer = timer_read();
        if (timer_elapsed(key_timer) < 100) {
          register_code(LCTL(KC_V));
        } else {
          register_code(LCTL(KC_C));
        }
        break;
      }
    return MACRO_NONE;
};
```

@mecanogrh , what is the advantage of doing that way instead?

mecanogrh, 2016-04-21:
Macros are called with M(number) you can't use LCTL(key_code) it obviously won't work :)

Benefit is if you have a bunch of keys to redefine you wrap all those in the function and don't write an timer_elapsed conditional for each one.

BeatVids, 2016-04-21:
What should I substitute (event.pressed) with then? O.o

mecanogrh, 2016-04-21:
Your are trying to assign two behaviours to a modifier+key?
If so the easiest way is to assign a layer to your control key then in this layer assign M(0) to the key (here C), then:

```c
static uint16_t key_timer;

const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt)
{
        case 0:
        if (event.pressed) {
          key_timer = timer_read();
          if (timer_elapsed(key_timer) < 100) {
            register_code(LCTL(KC_V));
          } else {
            register_code(LCTL(KC_C));
         }
          break;
        }
      }
    return MACRO_NONE;
};
```

BeatVids, 2016-04-21:
I sorta almost have it, the only thing is missing is getting to the copy part, I don't think I am getting the timer part correctly.

I have set M(1), omitting the original M(0)

```
static uint16_t key_timer;

const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt)
{
      switch(id) {
        case 1:
            if (record->event.pressed) {
              key_timer = timer_read();
              if (timer_elapsed(key_timer) < 100) {
                 return MACRO( D(LCTL), T(V), D(LCTL), END  );
              } else {
                 return MACRO( D(LCTL), T(C), D(LCTL), END  );
              }
            }
            break;
      }
    return MACRO_NONE;
};
```

eltang, 2016-04-21:
@BeatVids Here's the correct code.

```C
const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt) {
    switch(id) {
        case 1: {
            if (record->event.pressed) {
                key_timer = timer_read();
            }
            else {
                if (timer_elapsed(key_timer) < 100) {
                    return MACRO( D(LCTL), T(V), U(LCTL), END  );
                }
                else {
                    return MACRO( D(LCTL), T(C), U(LCTL), END  );
                }
            }
            break;
        }
    return MACRO_NONE;
}
```

BeatVids, 2016-04-21:
> `
>     record->event.pressed
> `
> This is a boolean value that can be tested to see if the switch is being pressed or released.

Aha, it was an either or situation, now I see the need for that else statement. Thank you very much everyone for your patience, and thanks @eltang for the final nail in the coffin.

Here is the entire snippet of code I have in case it helps any other newcomers in the future:

```
static uint16_t key_timer;

const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt)
{
  // MACRODOWN only works in this function
      switch(id) {
        case 0:
        if (record->event.pressed) {
           register_code(KC_RSFT);
        } else {
          unregister_code(KC_RSFT);
        }
        break;

        case 1:
            if () {
                key_timer = timer_read();
            }
            else {
                if (timer_elapsed(key_timer) < 100) {
                    return MACRO( D(LCTL), T(V), U(LCTL), END  );
                }
                else {
                    return MACRO( D(LCTL), T(C), U(LCTL), END  );
                }
            }
            break;

      }
    return MACRO_NONE;
};
```

closed 2016-04-21 by BeatVids

eltang, 2016-04-21:
My code should work as-is—you shouldn't need to change it!

BeatVids, 2016-04-21:
@eltang , I included the M(0) that comes with the default Ergodox keymap, in case anyone wanted to keep it! Sorry, I'll edit it!

eltang, 2016-04-21:
No worries! Having `M(0)` there is not a problem. It's `M(1)` that should be fixed.

mecanogrh, 2016-04-21:
```c
if()
```
works? if so it should be always true :)

BeatVids, 2016-04-22:
Sorry about that, I could have sworn I copied it correctly. I'm in the process from colemak, to colemak-dh if that allows me to blame it on something :laughing:

ezuk, 2016-04-22:
So what's the complete snippet? :) I'd like to add this as a section in the
docs.

eltang, 2016-04-22:
@ezuk That would be mine.

BeatVids, 2016-04-22:
Hi Erez!
Thanks to eltang, I added the "case 1" section to the keymap I've been editing from the default EZ colemak layout.
It allows a single button to copy and paste, just hold it a little, and boom, copied, and just tap to paste.

There's nothing else to it, besides setting the M(1) part.
I've also added the same feature to another button, but with undo/redo.
If you want my keymap.c, feel free to email me! However, it's still a work in progress, and I have no idea when I'll be complete with it!

ezuk, 2016-04-22:
Paste the keymap here! :) or in a gist. Way better than email. Love this
idea btw!

eltang, 2016-04-22:
@BeatVids You may want to invert the behavior of your copy/paste button. You have a visual indication of when something's been pasted, but not when it's been copied.

BeatVids, 2016-04-22:
ericramos1990

BeatVids, 2016-04-22:
Here is the link, let me know if I did anything wrong, I don't even know what a gist is really:
https://gist.github.com/BeatVids/65c2cdfb1d538611e681fd1fb3cd16f8

@eltang I thought about it, but holding to copy makes more sense for me. I see it as "charging up", and tapping to shoot! Plus, it would be easier to accidentally overwrite the clipboard with a tap, but to each his own!

mecanogrh, 2016-04-22:
@BeatVids I think you may have post a password here by mistake (e••••••••••••••0) though maybe not critical it would be a good idea to change it ;)

eltang, 2016-05-07:
@BeatVids Here is some code that will take up less space. Note that you will have to put `F(0)` in your main `keymaps` array and `ACTION_MACRO_TAP(0)` in your `fn_actions` array.

```c
case 0:
    if (record->event.pressed) {
        register_mods(MOD_BIT(KC_LCTL));
        if (record->tap.count) {
            register_code(KC_V);
            unregister_code(KC_V);
        }
        else {
            register_code(KC_C);
            unregister_code(KC_C);
        }
        unregister_mods(MOD_BIT(KC_LCTL));
    }
    break;
```

BeatVids, 2016-05-07:
That's awesome Eric, I wish I thought of that too.
It's a good feeling being clever when you share identical parts of code, the ctrl in this case :+1:

eltang, 2016-05-07:
@BeatVids I'm a bit of an optimization freak. :D Does it work?

BeatVids, 2016-05-07:
Would you be able to edit it to do this?
It copies with a slow hold, but cuts with a .5 second cut!

```
        case 2: {
            if (record->event.pressed) {
                key_timer = timer_read();
            }
            else {
                if (timer_elapsed(key_timer) < 150) {
                    return MACRO( D(LCTL), T(V), U(LCTL), END  );
                }
                else if (timer_elapsed(key_timer) > 500) {
                    return MACRO( D(LCTL), T(X), U(LCTL), END  );
                }
                else {
                    return MACRO( D(LCTL), T(C), U(LCTL), END  );
                }
            }
            break;
        }
```

eltang, 2016-05-07:
That's going to be a little harder, but I can make tap to copy, double tap to cut, and hold to paste!

BeatVids, 2016-05-07:
Hmm, I'm not really looking to double tap, because on some rare occasions, I will want to spam the paste function. But it will definitely help someone who doesn't want the paste feature. Thank you for the alternative, I will keep that in mind if I ever make a new function :)

eltang, 2016-05-08:
Actually, there's a small optimization that can be made to your current code. Have a look.

```c
case 0: {
    static uint16_t last_keypress_time;
    if (record->event.pressed) {
        last_keypress_time = record->event.time;
    }
    else {
        if (TIMER_DIFF_16(record->event.time, last_keypress_time) < 250) {
            return MACRO( D(LCTL), T(V), U(LCTL), END  );
        }
        else if (TIMER_DIFF_16(record->event.time, last_keypress_time) < 500) {
            return MACRO( D(LCTL), T(X), U(LCTL), END  );
        }
        else {
            return MACRO( D(LCTL), T(C), U(LCTL), END  );
        }
    }
    break;
}
```

BeatVids, 2016-05-10:
Saved it, thanks for the awesome alternative!

moonrumble, 2016-06-09:
What is the difference between function and macro? There are 2 solutions here for the Dual role key. One can use fn_action and don't need to use timer or we can purely using Macro but we need to use timer function.