qmk/qmk_firmware #281: CTL_T(LCTL(KC_A)) not working

jasontwong, 2016-04-23:
I'm trying to use the code: `CTL_T(LCTL(KC_A))` and it doesn't seem to work. The expected functionality would be to do `CTRL + A` on tap and `CTRL` on hold. Am I using it incorrectly?

eltang, 2016-04-23:
You can't chain those like that! You have to use a custom macro.

jasontwong, 2016-04-23:
I figured you couldn't since it says it takes a `kc`. I tried implementing the macro but only got the hold portion to work. Can anyone help?

```c
    case 0:
      if (record->event.pressed) {
        if (record->tap.count) {
          if (record->tap.interrupted) {
            record->tap.count = 0;
            // hold press action
            register_code(KC_LCTRL);
          } else {
            // tap press action
            return MACRO( D(LCTRL), T(A), U(LCTRL), END  );
          }
        } else {
          // hold press action
          register_code(KC_LCTRL);
        }
      } else {
        if (record->tap.count) {
          // tap release action
          return MACRO( D(LCTRL), T(A), U(LCTRL), END  );
        } else {
          // hold release action
          unregister_code(KC_LCTRL);
        }
        record->tap.count = 0;
      }
      break;
```

eltang, 2016-04-23:
Oops, I forgot to mention that you have to call the macro in a special way. You have to put `F(0)` in your `keymaps` array, put `[0] = ACTION_MACRO_TAP(0),` in your `fn_actions` array. also, notice that the keyboard shortcut will be executed twice if you tap the key to which this macro is bound.

sboesebeck, 2016-04-23:
I did similar things with a Macro using the timer functions, and, if you want to use the full keycode (KC_LCTRL) you can use the Macro-Function DOWN instead of D.

Then you can also use international keycodes, e.g. MACRO(DOWN(DE_OSX_AE),TYPE(KC_1),UP(DE_OSX_ACUT)) - you cannot combined keycodes there though. At least as I understand it. Like DE_OSX_QUOT will not work, as it is LSFT(DE_OSX_HASH).

The ErgodoxMacroGenerator actually generates macros for that purpose - at least for me, they work. Have a look at the osx_de and osx_de_experimental keymaps, hope that'll help...

cheers,

Stephan

jasontwong, 2016-04-24:
Thanks for all the help! I got everything working. I felt really limited when using the massdrop keyboard configurator. Now I'm overwhelmed with options! I'll make a PR to add my keymap.

closed 2016-04-24 by jasontwong

eltang, 2016-04-24:
What was the problem?

jasontwong, 2016-04-24:
Sorry, the solution was to use the `F(0)` combined with `ACTION_MACRO_TAP` in `fn_actions`. Working code below.

```c
const uint16_t PROGMEM fn_actions[] = {
  [0] = ACTION_MACRO_TAP(0)
};

const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt) // this is the function signature -- just copy/paste it into your keymap file as it is.
{
  switch(id) {
    case 0:
      if (record->event.pressed) {
        if (record->tap.count) {
          if (record->tap.interrupted) {
            record->tap.count = 0;
            // hold press action
            register_code(KC_LCTL);
          } else {
            // tap press action
            return MACRO( D(LCTL), T(A), U(LCTL), END  );
          }
        } else {
          // hold press action
          register_code(KC_LCTL);
        }
      } else {
        if (record->tap.count) {
          // tap release action
        } else {
          // hold release action
          unregister_code(KC_LCTL);
        }
        record->tap.count = 0;
      }
      break;
  }
  return MACRO_NONE;
};
```

eltang, 2016-04-24:
Awesome!

eltang, 2016-04-26:
@jasontwong I updated the code template.