qmk/qmk_firmware #288: Linux Hyper Key

SevereOverfl0w, 2016-04-26:
Linux has a native hyper key code. Would it be possible to reference this somehow from this firmware? I already have mappings in place using it, and it's somewhat simpler in places.

eltang, 2016-04-26:
Do you know which one it is? If you don't do you know the position of the key Linux recognizes as Hyper on a traditional keyboard?

SevereOverfl0w, 2016-04-26:
https://bbs.archlinux.org/viewtopic.php?pid=595002#p595002

Looks like it's e068? Or, escaped 0x68 (scancode)

jackhumbert, 2016-04-26:
Hyper is all on the OS's side - I don't think there's any way to map it natively, but you can stick in `KC_MENU` or something similar to be redefined in xmodmap or the equivalent for your distro.

jackhumbert, 2016-04-26:
Scancodes and keycodes are different, but you could try sticking in `0x68` (`KC_F13` I believe) to see what happens on your system.

SevereOverfl0w, 2016-04-26:
http://www.usb.org/developers/hidpage/Hut1_12v2.pdf Page ~58. No reference to hyper in here.

I've had a look through the file `/usr/include/linux/input.h` and it's include, and it doesn't look like there is any reference to hyper in there.

XFree86 keycode:
```
    // Other codes never generated.  The XFree86 ddx never generates
    // these codes.
    // Thus we can use them as fake keys
    <HYPR> =   128; // <U80>
```

So I guess it's a fake key implemented at one of the keyboard layers _after_ the fact. I think it's xfree86 (X.org) layer, but I'm not sure.

Regardless, this can be closed.

closed 2016-04-26 by SevereOverfl0w